cocotb/cocotb #4291: task cannot be terminated normally.

hhucchenyixiao, 2024-12-04:
cocotb version 1.8.1
Linux version 5.15.167.4-microsoft-standard-WSL2
Verilator 5.020 2024-01-01 rev (Debian 5.020-1)
python 3.12.3

I initiated two tasks concurrently, and when either of the tasks completes, both tasks will be terminated and enter the next loop iteration.

```python
for i in range(10):
  task1 = create_task(async_func_1(dut))
  task2 = create_task(async_func_2(dut))
  await First(task1, task2)
```

At the start of task1 and task2, I also `start_soon` some `log_chaned_signal` that print changing values.

```python
async def log_chaned_signal(dut, signal, signal_name=""):
  signal_old = None
  while True:
    await RisingEdge(dut.clk)
    if signal_old is None:
      signal_old = get_int(signal)
      signal._log.info(f"{signal_name}: {signal_old}")
    signal_new = get_int(signal)
    if (signal_new != signal_old):
      signal_old = signal_new
      signal._log.info(f"{signal_name}: {signal_new}")
```

However, as the loop progresses, I noticed that these printing tasks do not terminate with the completion of task1 and task2, so the same signal gets printed multiple times, for example:

```plaintext
18499.00ns INFO     cocotb.count                       count: 1
18499.00ns INFO     cocotb.count                       count: 1
18499.00ns INFO     cocotb.count                       count: 1
18499.00ns INFO     cocotb.count                       count: 1
18499.00ns INFO     cocotb.count                       count: 1
24486.00ns INFO     cocotb.count                       count: 2
24486.00ns INFO     cocotb.count                       count: 2
24486.00ns INFO     cocotb.count                       count: 2
24486.00ns INFO     cocotb.count                       count: 2
24486.00ns INFO     cocotb.count                       count: 2
```

It can be seen that at the same moment, multiple `log_chaned_signal` tasks are printing the signal value simultaneously.

I am seeking assistance to understand why these logging tasks are not stopping when task1 and task2 finish, and how I can ensure that each logging task stops appropriately after its corresponding main task has completed. Any advice on properly managing the lifecycle of these concurrent tasks would be greatly appreciated.

ktbarrett, 2024-12-05:
Not in cocotb 1.X. However, this will be possible in 2.0.

ktbarrett, 2024-12-05:
Since this is already on my radar and a known issue, I'll close this. Keep an eye out for cocotb 2.0!

closed 2024-12-05 by ktbarrett