cocotb/cocotb #4292: Simulation runs indefinitely after cocotb errors

parthkalgaonkar, 2024-12-04:
Basic info:
- cocotb version : 1.9.1
- OS version : RHEL7
- Simulator version : Xcelium 23.09-s008
- Python version : 3.9.2

If python fails to load (due to ImportError or some such issue), sometimes simulator can detect that nothing is to be done in the simulation and it ends immediately. However, if there is some activity in a sub-hierarchy, simulation keeps running indefinitely. In my particular case, a memory cell model has a free-running clock.

Here's an MVE

```sv
// test.sv
`timescale 1ns/1ps

module test(
    input logic     clk,
    input logic     d,
    output logic    q
);

    always @(posedge clk) begin
        q <= #1ns d;
    end

    sub_test m_sub(
        .d(d)
    );
endmodule

module sub_test(
    input logic d
    );
    logic   m_clk;

    initial begin
        m_clk = 0;
        forever begin
            #1ns;
            m_clk = ~m_clk;
        end
    end

    initial begin
        while (1) begin
            repeat (100) begin
                @(posedge m_clk);
            end
            $display("%t: d=%d", $time, d);
        end
    end

    initial begin
        #1us $finish();     // without this simulation would never stop
    end
endmodule
```

```py
# test.py
import cocotb
from cocotb.tribbers import Timer   # Intentionally added ModuleNotFoundError

@cocotb.test()
async def m_test(dut):
    await Timer(10, "ns")
    dut.d.value = 1
    await Timer(400, "ns")
    dut.d.value = 0
```

```py
# test_runner.py
from pathlib import Path
from cocotb.runner import get_runner

def test_all():
    sim = "xcelium"  # also tried with iVerilog
    proj_path = Path(__file__).resolve().parent

    sources = [
            proj_path / "test.sv"
            ]

    runner = get_runner(sim)
    runner.build(sources=sources, hdl_toplevel="test")
    runner.test(
            hdl_toplevel="test",
            test_module="test",
            log_file="sim.log",
            verbose=True
            )

if __name__ == "__main__":
    test_all()
```

[test.py.txt](https://github.com/user-attachments/files/18007916/test.py.txt)
[test.sv.txt](https://github.com/user-attachments/files/18007917/test.sv.txt)
[test_runner.py.txt](https://github.com/user-attachments/files/18007920/test_runner.py.txt)

When running this, cocotb fails with the `Passing event to upper layer failed` gpi error and then simulation keeps running

```
	Processing -LOADVPI request(s)
		Bootstrap:  vlog_startup_routines_bootstrap()  library:  [redacted]/python3.9/site-packages/cocotb/libs/libcocotbvpi_ius.so
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at [redacted]/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered

xcelium> source [redacted]
xcelium> run; exit;
     0.00ns INFO     cocotb                             Running on xmsim(64) version 23.09-s008
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.1 from [redacted]/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1732716421
     0.00ns CRITICAL cocotb.regression                  Failed to import module test: No module named 'cocotb.tribbers'
     0.00ns INFO     cocotb.regression                  MODULE variable was "test"
     0.00ns INFO     cocotb.regression                  Traceback:
     0.00ns INFO     cocotb.regression                  Traceback (most recent call last):
                                                          File "[redacted]/python3.9/site-packages/cocotb/regression.py", line 203, in _discover_tests
                                                            module = _my_import(module_name)
                                                          File "[redacted]/python3.9/site-packages/cocotb/regression.py", line 72, in _my_import
                                                            mod = __import__(name)
                                                          File "[redacted]/python3.9/site-packages/_pytest/assertion/rewrite.py", line 178, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "[redacted]/test.py", line 2, in <module>
                                                            from cocotb.tribbers import Timer
                                                        ModuleNotFoundError: No module named 'cocotb.tribbers'

Traceback (most recent call last):
  File "[redacted]/python3.9/site-packages/cocotb/__init__.py", line 233, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "[redacted]/python3.9/site-packages/cocotb/__init__.py", line 316, in _initialise_testbench_
    regression_manager = RegressionManager.from_discovery(top)
  File "[redacted]/python3.9/site-packages/cocotb/regression.py", line 174, in from_discovery
    return cls(dut, tests)
  File "[redacted]/python3.9/site-packages/cocotb/regression.py", line 153, in __init__
    for test in tests:
  File "[redacted]/python3.9/site-packages/cocotb/regression.py", line 203, in _discover_tests
    module = _my_import(module_name)
  File "[redacted]/python3.9/site-packages/cocotb/regression.py", line 72, in _my_import
    mod = __import__(name)
  File "[redacted]/python3.9/site-packages/_pytest/assertion/rewrite.py", line 178, in exec_module
    exec(co, module.__dict__)
  File "[redacted]/test.py", line 2, in <module>
    from cocotb.tribbers import Timer
ModuleNotFoundError: No module named 'cocotb.tribbers'
Traceback (most recent call last):
  File "[redacted]/python3.9/site-packages/cocotb/__init__.py", line 332, in _sim_event
    scheduler.log.error(msg)
AttributeError: 'NoneType' object has no attribute 'log'
     0.00ns ERROR    gpi                                Passing event to upper layer failed
                 199: d=z
                 399: d=z
                 599: d=z
                 799: d=z
                 999: d=z

	PLIVERBOSE API stats
		Handle counts
		   PLI/VPI handles:  1
xmsim: Memory Usage - Final: 73.4M, Peak: 76.0M, Peak virtual: 225.0M
xmsim: CPU Usage - 0.1s system + 0.4s user = 0.4s total (22.4s, 1.8% cpu)
TOOL:	xrun(64)	23.09-s008: Exiting on Nov 27, 2024 at 19:37:04 IST  (total: 00:00:22)
--- ncdbg_exit() has been called
```

Would be nice if this kind of errors can be caught and simulations end immediately. I don't see why someone would want the simulation to continue if the testcase has not been loaded at all.

ktbarrett, 2024-12-05:
Hmmmm... I thought we had fixed this issue, but I suppose not. Thanks for the report.

ktbarrett, 2024-12-22:
Fixed by #4310.

marlonjames, 2025-01-03:
xref #2018

PR #4310 (2025-01-13): More Refactoring
Fixes #4292. Fixes #2018.

Removes the need to keep `_filter_from_c`, `_log_from_c`, and `_sim_event` at the module level in the PYGPI_ENTRY_POINT. This allows us to support multiple entry points, which means we can load things *before* the `cocotb` module is loaded. The primary intent of this is to support loading library coverage before the `cocotb` module is loaded, allowing us to collect more code coverage.

cocotb initialization error handling was also refactored. See that commit message for more details.

## TODO
- [x] newsfragment on change to PYGPI_ENTRY_POINT
closed 2025-01-13 by ktbarrett