cocotb/cocotb #4295: Cannot generate waveform.ghw file for VHDL sources with python test runner

m-tosch, 2024-12-09:
Info
* cocotb version: 1.9.2,
* OS: WSL2 (Ubuntu 20.04 64-bit) on a Windows 10 host
* Simulator: GHDL
* Python: 3.8.10 (self installed)

Hi, first time cocotb user here. Love it so far!

I am looking into how to use **python test runners** instead of **Makefiles**. As far as i understand python test runners are equivalent but experimental in cocotb 1.9.2.

I have a single VHDL source and a single test for it. With my Makefile i can generate a waveform file "waveform.ghw" and view it in gtkwave. This works perfectly. 

When i try to generate a waveform file via python test runner i am unable to. I have looked through the [runner API documentation](https://docs.cocotb.org/en/v1.9.2/library_reference.html#api-runner) and thought **plusargs** is what i need. 

**my_design.vhd**
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity my_design is
    generic(
        G_GENERIC : integer range 0 to 63
    );
    port (
        clk : in std_logic;
        reset : in std_logic
    );
end entity;

architecture Behavioral of my_design is

    signal s_signal_1 : std_logic := 'X'; -- uninitialized state
    signal s_signal_2 : std_logic := '0'; -- initialized to logic 0

begin

end architecture;
```

**test_my_design.py**
```python

import cocotb
from cocotb.triggers import RisingEdge, FallingEdge, Timer
from cocotb.clock import Clock

async def run_reset_routine(dut):
    for _ in range(3):
        await RisingEdge(dut.clk)
    dut.reset.value = 0

@cocotb.test()
async def my_test_case(dut):

    # Generate a clock
    cocotb.start_soon(Clock(dut.clk, 1, units="ns").start())

    # Reset DUT
    dut.reset.value = 1

    # Reset the module, wait 3 rising edges then release reset
    cocotb.start_soon(run_reset_routine(dut))

    await Timer(10, units="ns")  # wait a bit
    await FallingEdge(dut.clk)  # wait for falling edge/"negedge"

    expected = 16
    dut._log.debug(f"G_GENERIC is  {dut.G_GENERIC.value} in binary and {dut.G_GENERIC.value.integer} as integer")
    assert dut.G_GENERIC.value.integer == expected, f"Error: Generic is not {expected}!"
```

**Makefile**
```cmd
SIM ?= ghdl
TOPLEVEL_LANG ?= vhdl

VHDL_SOURCES += $(PWD)/*.vhd

TOPLEVEL = my_design

MODULE = test_my_design

SIM_ARGS = --wave=waveform.ghw

SIM_ARGS += -gG_GENERIC=16

include $(shell cocotb-config --makefiles)/Makefile.sim
```

**test_runner.py**
```python
import os
from pathlib import Path
from cocotb.runner import get_runner

def test_my_design_runner():
    sim = os.getenv("SIM", "ghdl")

    proj_path = Path(__file__).resolve().parent

    runner = get_runner(sim)

    hdl_toplevel = "my_design"

    runner.build(
        sources=[proj_path / "my_design.vhd"],
        hdl_toplevel=hdl_toplevel,
        parameters={
            "G_GENERIC": 16
        },
        waves=True,
    )

    runner.test(
        test_module="test_my_design,",
        hdl_toplevel=hdl_toplevel,
        hdl_toplevel_lang="vhdl",
        plusargs=["--wave=waveform.ghw"],
        waves=True,
    )

if __name__ == "__main__":
    test_my_design_runner()
```

The GHDL commands that are being executed look good to me with regards to argument placement for GHDL...

From Makefile
```cmd
 /usr/local/bin/ghdl -r    --time-resolution=ps --workdir=sim_build -Psim_build --work=work my_design  --vpi=/home/mto/.local/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --wave=waveform.ghw -gG_GENERIC=16  
```
From Python test runner
```cmd
INFO: Running command ghdl -i --work=top /home/mto/cocotb/my_design.vhd in directory /home/mto/cocotb/sim_build
INFO: Running command ghdl -m --work=top my_design in directory /home/mto/cocotb/sim_build
INFO: Running command ghdl -r --work=top my_design --vpi=/home/mto/.local/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --wave=waveform.ghw -gG_GENERIC=16 in directory /home/mto/cocotb/sim_build

```

ktbarrett, 2024-12-09:
Looks like this wasn't implemented for the GHDL runner yet. Pull Requests are welcome!

m-tosch, 2024-12-10:
**Turns out the waveform file generation via ghdl is already working with the python test runner.**

The only difference is that when a Makefile is used, waveform.ghw can be found in the same directory while with the python test runner the output will be captured in the sim_build folder

```
sim_build/waveform.ghw
```

Sorry for the confusion!

closed 2024-12-10 by m-tosch