cocotb/cocotb #4311: VCS Illegal VPI Handles 79/83

ipburbank, 2024-12-22:
Cocotb log contains dozens of these prints before testing actually starts:

```
     0.00ns ERROR    gpi                                VPI error
     0.00ns ERROR    gpi                                
                                                        Error-[VPI-ITPTVOT] Illegal VPI tag
                                                        Unknown source file, unknown line number
                                                          In VPI call vpi_handle, illegal type '79' was passed in for vpi object of 
                                                          type 'vpiLogicVar'
                                                          Please fix the VPI call in your PLI program by providing the legal type.
     0.00ns ERROR    gpi                                VPI error
     0.00ns ERROR    gpi                                
                                                        Error-[VPI-ITPTVOT] Illegal VPI tag
                                                        Unknown source file, unknown line number
                                                          In VPI call vpi_handle, illegal type '83' was passed in for vpi object of 
                                                          type 'vpiLogicVar'
                                                          Please fix the VPI call in your PLI program by providing the legal type.
     0.00ns WARNING  gpi                                VPI: Cannot discover range bounds, guessing based on elements
```

## Versions
Cocotb:
5d0f4f7677858c611a3de212d6c978fdd939b925

Ubuntu:

% lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy

VCS: vcs/U-2023.03

Python: Python 3.6 (using the python bundled with VCS)

ktbarrett, 2024-12-22:
What is the type of the variable? Is is a scalar logic object, or a logic array? The former would be invalid and there are plans in cocotb to split the LogicObject into a scalar and Array type which should prevent the issue. If its complaining about logic arrays that's a VCS issue.

ipburbank, 2024-12-22:
Minimal example:
```
module my_module (
    input logic clk
)

endmodule
```

and

```
async def test_my_moduledut)
    c = Clock(dut.clk, 10, "ns")
    await cocotb.start(c.start())
```

ktbarrett, 2024-12-22:
Ah yeah. The issue is simply that it's trying to get the range of a scalar. I'll keep this around. I plan on working on the feature soon. I have a few things in flight.

ipburbank, 2025-01-14:
Confirmed locally on my project that the warnings are gone. Thanks @ktbarrett!

closed 2025-01-14 by ipburbank