cocotb/cocotb #4313: Add documentation on remote debugging with `debugpy`

ktbarrett, 2024-12-22:
Microsoft's [`debugpy`](https://github.com/microsoft/debugpy) is a good candidate for debugging cocotb sims remotely. That can be useful for debugging non-terminal applications like most commercial simulations.

marlonjames, 2025-01-03:
I posted some info about how I use `debugpy` here: https://github.com/cocotb/cocotb/discussions/2652#discussioncomment-1135765

Zosili, 2026-11-12:
﻿I'd like to work on this.

I have a draft ready that adds a subsection to *Attaching a Debugger → Python*:

- `pip install debugpy`, then `debugpy.listen(("127.0.0.1", 5678))` and `debugpy.wait_for_client()` in the test module, optionally guarded by an environment variable so the test still runs without a debugger;
- attaching Visual Studio Code through a `launch.json` entry, including `pathMappings` for simulations running in a container or on a remote host;
- `debugpy.breakpoint()` for breakpoints that are ignored when no client is attached.

It also notes that `python -m debugpy` cannot be used to launch the simulation, because the Python interpreter is embedded in the simulator process.

Could you assign this issue to me so I can open the PR?

Zosili, 2026-11-12:
﻿Withdrawing my request for assignment: #5642 already implements this and is still open, so I am not going to duplicate it.