cocotb/cocotb #4322: Subscritable inout

davekeeshan, 2024-12-27:
Hi there

I am trying to address a subscripting of a vector problem over at [cocotb-bus](https://github.com/cocotb/cocotb-bus), [#86](https://github.com/cocotb/cocotb-bus/issues/86).  I wanted to assign a bus, like a jtag, to an inout vector, digilent for example usually, by default, expose their pmod pins as an 8 bit inout vector at the top level.

    inout [7:0] ja

(I know I can bit burst the vector at the top levell and mapp the vector to get each pin on its own, but that is not the point of this issue, it is to report and issue with when it is a vector.)    

I got it working and was able, in `cocotb`, see in signals and out signals working on the same vector, using the released version of `cocotb`, 1.9.2 at the time of writing.

However when I bump cocotb to the head of git, I get this complaint for the same code:

    TypeError: 'LogicObject' object is not subscriptable

I checked the issues and this was reported, #4274, as known but 'not planned' (not sure what that means, won't fix?)

I wrote up a test for cocotb-bus, which can be found here, [test_vector_assign](https://github.com/daxzio/cocotb-bus/tree/vector_assign/tests/test_cases/test_vector_assign). I get the `inout` to be an `in` or an `out` by using a an `iobuf` and hard coded the `T` pin in the RTL (all captured in the test)

In it I try and assign a and gate, `in a, b`, `out c` to a vector `ja[3:0]`, where 

    ja[0] => a
    jb[1] => b
    jb[2] => c

I looked at the suggested fix (slightly rewritten):

    temp = dut.ja.value
    ja[1] = 1
    dut.ja.value = temp

I don't believe this fix works as the last line appears to be setting the whole vector to something, however as c is an output, j[2] has a driver already and this code causes a conflict as it ends up getting driven by two sources.

I have this working under 1.9.2 but it is not working under 2.0-dev, or am I missing something

    cocotb: 1.9.2 - 2.0-dev
    OS: Linux 64
    SIM: Icarus
    python: 3.12.3 - spack

ktbarrett, 2024-12-27:
You are trying to index into a packed vector, which is not legal. I suggest making the inout declaration use an unpacked array instead. This is out of cocotb's hands as simulators inconsistently allow indexing into packed objects and tend to not work as expected when it does allow you. This is wontfix.

closed 2024-12-27 by ktbarrett

davekeeshan, 2024-12-27:
@ktbarrett pardon my pedantry, but I think it is important to capture the issue here, as it may be helpful for future scholars (which more than likely will just be me)

Where is it stated that it is not legal? Is it in the verilog LRM? or the VHDL equivalent. 

I am sure it is legal in the language, but not supported by all the tools which is different than not being legal.  I fight unsupported, but legal features, all the time between simulation and synthesis tools.

ktbarrett, 2024-12-27:
In VHDL is legal, but we don't allow it for consistentcy. In Verilog it's not *explicitly* disallowed, but is implicitly. TBD on the Verilog LRM sections.

imphil, 2024-12-28:
It would be nice to add this to the documentation (I guess in general we're not really good at documenting what data type cocotb returns for which VHDL/Verilog constructs). @davekeeshan A pull request would be very welcome here!

Maybe we can also make the error message more helpful, by detecting if people are trying to access a packed array?

ktbarrett, 2024-12-28:
I opened up #4323.