cocotb/cocotb #4325: test_custom_entry fails VCS with Segmentation fault

themperek, 2025-01-01:
Error:

```
Compiler version W-2024.09_Full64; Runtime version W-2024.09_Full64;  Jan  1 10:28 2025
     -.--ns ERROR    gpi                                ..mbed/gpi_embed.cpp:68   in get_interpreter_path            PYGPI_PYTHON_BIN variable not set. Can't initialize Python interpreter!
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered

An unexpected termination has occurred in sim_build/simv due to a signal: Segmentation fault

During a VPI callback function for callback reason="cbStartOfSimulation" 

Command line: sim_build/simv
```

ktbarrett, 2025-01-01:
Looks like PYGPI_PYTHON_BIN isn't set, initialization fails, then the shutdown callback is called on unintialized state causing a segfault?

themperek, 2025-01-01:
> Looks like PYGPI_PYTHON_BIN isn't set, initialization fails, then the shutdown callback is called on unintialized state causing a segfault?

Sorry wrong logfile. Is updated.

themperek, 2025-01-01:
Looks like my mistake here.

closed 2025-01-01 by themperek