cocotb/cocotb #4326: test_timeout_testdec_simultaneous fails in VCS

themperek, 2025-01-01:
`test_timeout_testdec_simultaneous ` in `tests/test_cases/test_cocotb/test_tests.py`

Fails with assert:
```python
assert False, "Expected a Timeout"
```

themperek, 2025-01-01:
It works if uses  `Timer(2, "ns")`.
```
async def test_timeout_testdec_simultaneous(dut):
    try:
        await cocotb.triggers.with_timeout(
            Timer(2, "ns"), timeout_time=1, timeout_unit="ns"
        )
    except SimTimeoutError:
        pass
    else:
        assert False, "Expected a Timeout"
 ```
 
#4324 
 @ktbarrett Something with exact order if TImer is `1ns` and timeout exactly `1ns` what is first?

ktbarrett, 2025-01-01:
They should be de-duplicated and there only be one actual VPI callback registered. I wonder if there's a bug and `_prime` is being called more than once.

themperek, 2025-01-02:
What will be the print order? Is it deterministic?

```python3

@cocotb.test()
async def test(dut):

    async def coro():
        await Timer(1, "ms")
        print("coro")

    await cocotb.start(coro())
    await Timer(1, "ms")
    print("Timer")

    await Timer(1, "ms")

```

Icarus:
```
coro
Timer
```

VCS:
```
Timer
coro
```

ktbarrett, 2025-01-02:
I misspoke actually. `Timer` is not de-duplicated, multiple GPI triggers are registered. There is no guarantee about order of evaluation of GPI callbacks. The test makes a faulty assumption, I'd just remove it.