cocotb/cocotb #4333: combine_results.py does not raise and exit 1 if there has been a dependencay failure

davekeeshan, 2025-01-03:


ktbarrett, 2025-01-03:
Dependency failure?

davekeeshan, 2025-01-03:
> Dependency failure?

Apologies my computer posted before I had completed the report, should be in place now

davekeeshan, 2025-01-03:
OK the PR I opened has errors every where, #4334, which suggests that `results.xml` is not the best thing to use, maybe something else is appropriate. Hopefully the scope of the error is clear from the passing/failing run above.

I have made that change to my own flow, it highlighted some further errors with some of my work and helped me fix them.

https://github.com/daxzio/cocotbext-jtag/actions/runs/12600842340/job/35120736730

themperek, 2025-01-04:
> Here is an example of a passing failing test:
> 
> https://github.com/daxzio/cocotbext-jtag/actions/runs/12600538398/job/35119795921

Makefile failed (https://github.com/daxzio/cocotbext-jtag/actions/runs/12600538398/job/35119795921#step:9:79) . Should return error and fail the test. Why does  it pass? Looks like something with your setup?

davekeeshan, 2025-01-04:
This was discussed over on gitter a while back, but because python runs wholly inside the simulator, an exit event inside python is not guaranteed to be passed out of the simulator to be picked up by the CI. 

It is my understanding that is why `combine_results.py` exists, to collate and return the passes and escalate the fails. In its current incarnation it seems only sensitive to the report generated at the end and is not reporting catastrophic fails, like dependancy conflicts, which can occur on initial setup (trying to cover multiple versions of python) or cocotb migration where fundamental features change. For me at least this is a major reason to use a CI. 

My setup is visible, if you think something is missing I of course would love to get it right and fixed. My fix, at least, highlights and solves the problem I have, so in some ways I am safe to move on, however I felt this might be something the community might also be seeing.

davekeeshan, 2025-01-05:
> Makefile failed (https://github.com/daxzio/cocotbext-jtag/actions/runs/12600538398/job/35119795921#step:9:79) . Should return error and fail the test. Why does it pass? Looks like something with [...]

OK I didn't undestand this correctly at the time (following up bugs while shopping is a no-no).

What you are saying is that an exit event is coming from cocotb, because of the failure.  However because I am doing this:

    cd tests/test001 ; make clean sim ; ../../rtlflo/combine_results.py
    
calling the follow on script in a single line erases the exit event, from the CI/Shell perspective, only the output of the `combine_results.py` is significant.

I think to catch the makefile fail, it needs to be on its own line, and the combine_results.py on a separate one (which the cd-oing into the directory for both, obviously less elegant, not DRY):

    cd tests/test001 ; make clean sim
    cd tests/test001 ; ../../rtlflo/combine_results.py

ktbarrett, 2025-01-05:
@davekeeshan I see what you are getting at. No tests are collected before the failure, the results.xml is written but there are no "failures" because there are no tests, and thus it "passes". Is that what you are seeing?

davekeeshan, 2025-01-05:
@ktbarrett Yes i still struggle with Makefiles after all these years (though I think that hack is a bash one) that is the correct one-line, DRY answer, I think that addresses the concern I was having.  Thanks for the input and the patience, i know this will be of use to future users. Closing