cocotb/cocotb #4340: Separate `cocotb.triggers` into base triggers and extended triggers

ktbarrett, 2025-01-05:
Right now there is a cyclic dependency between `cocotb.task` and `cocotb.triggers` because many of the "extended" triggers work on Tasks. Separating them could look like this...

```
# triggers.py
# no dependencies
_CallbackHandle
Trigger
Event
Lock
NullTrigger

# gpi_triggers.py
# depends on triggers.py
GPITrigger + friends
ClockCycles

# task.py
# depends on triggers.py
Join
Task

# extended_triggers.py
# depends on triggers.py and task.py
First
Combine
with_timeout
```

And we are left with a tree of dependencies with no cycles.

marlonjames, 2025-01-08:
Would we have a `base_triggers.py` or something and then just import everything into `triggers.py` to keep the namespacing backwards-compatible?

ktbarrett, 2025-02-13:
Closed by #4451.

closed 2025-02-13 by ktbarrett