cocotb/cocotb #4343: Crash with cocotb + latest scipy 1.15

andrewpeck, 2025-01-06:
Dear all, 

I recently ran into an issue with cocotb when importing scipy with the recently released `v1.15.0` (released last Friday).

```
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/_pytest/assertion/rewrite.py", line 184, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/scipy/ndimage/_support_alternative_backends.py", line 65, in <module>
                                                            delegator = getattr(_delegators, func_name + "_signature")
                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

The issue can be reproduced very simply... in any cocotb test bench just add `from scipy import *`.

I tried doing this in several test cases in cocotb, both with Make based runners as well as the python runner.

I am using these tool versions

```
cocotb=1.9.2 
os=linux 64 bit 
simulator=verilator 5.020
python=3.12.3
```

An issue was reported to scipy, https://github.com/scipy/scipy/issues/22236 but it is not clear that it is a scipy bug itself. I don't understand any of the details with assertion rewriting in cocotb so my effort to debug further stopped here.. it is not clear at all to me if the fix for this issue needs to come from scipy or pytest or cocotb and am hoping somebody here with better understanding could shed some light on this.

I for now have pinned scipy to the previous version and can continue working.

Thank you in advance for any help and for all your work on this excellent tool.

A full trace is included below.

```
-------------------------------------------------------------------------------------------------------- Captured stdout call --------------------------------------------------------------------------------------------------------
INFO: Running command perl /usr/local/bin/verilator -cc --exe -Mdir /home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1 -DCOCOTB_SIM=1 --top-module runner --vpi --public-flat-rw --prefix Vtop -o runner -LDFLAGS '-Wl,-rpath,/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/cocotb/libs -L/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/cocotb/libs -lcocotbvpi_verilator' -DDEFINE=4 -I/home/apeck/work/cocotb/tests/designs/basic_hierarchy_module -GWIDTH_IN=16 /home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/cocotb/share/lib/verilator/verilator.cpp /home/apeck/work/cocotb/tests/designs/runner/runner.v in directory /home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1
INFO: Running command make -C /home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1 -f Vtop.mk VM_TRACE=0 in directory /home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1
make: Entering directory '/home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1'
make: Nothing to be done for 'default'.
make: Leaving directory '/home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1'
WARNING: pre_cmd is not implemented for Verilator.
INFO: Running command /home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1/runner in directory /home/apeck/work/cocotb/tests/pytest/sim_build/test_runner/WIDTH_IN=16_pre_cmd1
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/apeck/work/cocotb/.venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.020 2024-01-01
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.2 from /home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1736182940
     0.00ns CRITICAL cocotb.regression                  Failed to import module test_runner: module 'scipy.ndimage._delegators' has no attribute '@py_builtins_signature'
     0.00ns INFO     cocotb.regression                  MODULE variable was "test_runner"
     0.00ns INFO     cocotb.regression                  Traceback (most recent call last):
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/cocotb/regression.py", line 203, in _discover_tests
                                                            module = _my_import(module_name)
                                                                     ^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/cocotb/regression.py", line 72, in _my_import
                                                            mod = __import__(name)
                                                                  ^^^^^^^^^^^^^^^^
                                                          File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
                                                          File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
                                                          File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/_pytest/assertion/rewrite.py", line 184, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/apeck/work/cocotb/tests/pytest/test_runner.py", line 9, in <module>
                                                            from scipy import *
                                                          File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
                                                          File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
                                                          File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/_pytest/assertion/rewrite.py", line 184, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/scipy/ndimage/__init__.py", line 156, in <module>
                                                            from ._support_alternative_backends import *
                                                          File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
                                                          File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
                                                          File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/_pytest/assertion/rewrite.py", line 184, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/apeck/work/cocotb/.venv/lib/python3.12/site-packages/scipy/ndimage/_support_alternative_backends.py", line 65, in <module>
                                                            delegator = getattr(_delegators, func_name + "_signature")
                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                        AttributeError: module 'scipy.ndimage._delegators' has no attribute '@py_builtins_signature'. Did you mean: 'grey_dilation_signature'?
                                                        
     0.00ns ERROR    cocotb                             module 'scipy.ndimage._delegators' has no attribute '@py_builtins_signature'
- :0: Verilog $finish
```

ktbarrett, 2025-01-06:
Looks like pytest assertion rewriting and scipy don't work well together. I wonder if there is an upstream issue about this? Alternatively we would need to introduce a switch to turn this off.

andrewpeck, 2025-01-06:
I've only found the linked issue reported to SciPy https://github.com/scipy/scipy/issues/22236

ktbarrett, 2025-01-06:
So it looks like scipy decided it's not their problem because pytest assertion rewriting shouldn't be applied to their sources. The issue being that pytest's `python_files` option we use to select which sources to enable rewriting on doesn't support negative patterns, so we can't say `!scipy/**/*.py` or similar.

ktbarrett, 2025-01-06:
I think for now we are going to have to add a switch to turn off assertion rewriting in its entirety until pytest adds this option.

cmarqu, 2025-01-06:
Based on just reading https://github.com/pytest-dev/pytest/issues/10845, could it be that a workaround is `PYTEST_PLAIN_ASSERTS=true PYTEST_ADDOPTS="--assert=plain" <cocotb invocation>`?

andrewpeck, 2025-01-06:
Thanks for looking into this and filing upstream issue @ktbarrett 

@cmarqu I just tried `PYTEST_PLAIN_ASSERTS=true PYTEST_ADDOPTS="--assert=plain" make` on my testcase but see the same issue.

ktbarrett, 2025-06-05:
@cmarqu I think those environment variables are only used if you enter pytest through it's `main`. I think the only think we can do is give the option to turn the thing off wholesale and hope projects like `scipy` have the decency to prevent the issue since `pytest` refuses. I'll try to get this into 2.0.

andrewpeck, 2025-06-05:
> [@andrewpeck](https://github.com/andrewpeck) is this issue resolved by `scipy==1.15.3`?

thanks @anton-buyskikh  it seems to do the trick!

PR #4728 (2025-06-10): Add envvar to customize installing pytest assertion rewriting hook
Closes #4343. Applying rewriting to all Python files has some issues, so we add this option to allow users to select which files to rewrite, or to disable it entirely.
closed 2025-06-10 by ktbarrett