cocotb/cocotb #4359: Please switch from setup-ghdl-ci to setup-ghdl

Paebbels, 2025-01-12:
> [!CAUTION]
> This GitHub Action (`setup-ghdl-ci`) has been **deprecated** in favor of the new [setup-ghdl](https://github.com/ghdl/setup-ghdl) Action.
> Please update your YAML files and transition to the new GitHub Action. See the linked repository for details on optional parameters.

**Quick example:**
```yml
jobs:
  GHDL-on-Ubuntu:
    runs-on: ubuntu-24.04
    steps:
      - name: Run VHDL Simulation
        uses: ghdl/setup-ghdl@v1
        with:
          version: nightly
          backend: mcode
```

**Why has it been changed?**  
* The Javascript GitHub Action is hard to maintain.  
  In addition, it has a high complexity with Node.js and other tools compare to a much simpler Composite Action using simple shell instructions in Bash/Powershell.
* This action doesn't support the `ubuntu-24.04` (= `ubuntu-latest`) runner image.
* This Action uses wrong tagging to select GHDL versions. See analysis: https://github.com/ghdl/ghdl/issues/2852
* Due to changes in GHDL's nightly releases and attached assets, the filenames are broken.

-----------------------
This repository has been listed by [Dependabot](https://github.com/ghdl/setup-ghdl-ci/network/dependents) as a dependent of `setup-ghdl-ci`.

ktbarrett, 2025-01-12:
We do not use this action. @Paebbels If possible you might want to check the dependencies of projects marked as dependents. I've found the dependency resolution to be pretty unreliable.

closed 2025-01-12 by ktbarrett

Paebbels, 2025-01-12:
Ok, thanks. I did filtering to remove university lab repositories and inactive repositories and from remaining 15+ repositories. I then quickly checked 3 by hand. They used it .).

At least, I didn't broke cocotb with my updates a few weeks ago - this also explains why cocotb didn't create an issue at GHDL ;).