cocotb/cocotb #4360: dump waves with Quickstart example

zhqhzj, 2025-01-13:
[image]

zhqhzj, 2025-01-13:
so, I can't gen *.vcd file

zhqhzj, 2025-01-13:
And other errors.
[image]

themperek, 2025-01-13:
- "No tests were discovered" -> there is something wrong with your Makefile, python module 
- Icarus 10.2 -> is quite an old version and may not support some features
- `No module cocotb_tools` -> it seems you have  some mixture of cocotb 2.0 and 1.9 in your environment

zhqhzj, 2025-01-13:
[root@eda aaa]# yum install iverilog
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirrors.aliyun.com
 * centos-sclo-rh: mirrors.aliyun.com
 * centos-sclo-sclo: mirrors.aliyun.com
 * epel: d2lzkl7pfhq30w.cloudfront.net
 * extras: mirrors.aliyun.com
 * updates: mirrors.aliyun.com
Package iverilog-10_2-2.el7.x86_64 already installed and latest version
Nothing to do

i'll try update the tool later

But i use xcelium dump *shm file, all pass!!!   My fifo.sv case

closed 2025-01-13 by zhqhzj