cocotb/cocotb #4387: Can't pass --error-limit 1000 option to Verilator using runner - spaces in argument seem to break the command line

rbarzic, 2025-01-23:
While trying to add --error-limit to the command line of Verilator to try to debug some other issue,  I noticed that the verilator command line seems to incorrectly contains '    '.

I use the following code 
```python
 sim = os.getenv("SIM", "icarus")
        if sim == "verilator":
            build_args = ["--timing", "--error-limit 1000"]
        else:
            build_args = []
        self.runner = get_runner(sim)
        self.defines.update(extra_defines)
        self.runner.build(
            verilog_sources=self.verilog_sources,
            vhdl_sources=[],
            hdl_toplevel=self.toplevel,
            always=True,
            defines=self.defines,
            includes=self.includes,
            verbose=True,
            build_args = build_args
        )
```

I got the error 
----------------------------- Captured stderr call -----------------------------
%Error: Invalid option: --error-limit 1000... Suggested alternative: '-error-limit'

From the log I have this call to verilator - Note the ' ' around --error-limit 1000
INFO: Running command perl /opt/verilator/2025-01-23-0507fb46//bin/verilator -cc --exe -Mdir /home/roba/work/gitlab/icdesign/ips/top_ulv/sim/cocotb_wrapper2/sim_build -DCOCOTB_SIM=1 --top-module chip --vpi --public-flat-rw --prefix Vtop -o chip -LDFLAGS '-Wl,-rpath,/home/roba/.virtualenvs/top_ulv2/lib/python3.10/site-packages/cocotb/libs -L/home/roba/.virtualenvs/top_ulv2/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator' --timing '--error-limit 1000' -DCOCOTB_SIM=1 -DCHIP_COCOTB_VCD=1 

I've tried verilator separately:
verilator '--error-limit 100' --binary ../../rtl/verilog/ulvregif_apbif.v -I../../include/verilog
Got:
%Error: Invalid option: --error-limit 100... Suggested alternative: '-error-limit'

I noticed a "similar" issue, if I use 
 build_args = [" --timing "]   # note the extra spaces
instead of 
 build_args = ["--timing"]

 verilator --version :
Verilator 5.033 devel rev v5.032-42-g0507fb465

ktbarrett, 2025-01-23:
You'd think option parsing would be a solved problem in 2025. I don't think cocotb should be duplicating the option parsing that Verilator is also doing, so I'd suggest this is an upstream issue.

As for a workaround. Have you tried splitting the argument into two elements, or using the syntax `--error-limit=1000`?

imphil, 2025-01-29:
> As for a workaround. Have you tried splitting the argument into two elements (e.g. ["--error-limit", "1000"]), or using the syntax --error-limit=1000?

I'd say that's not a workaround, but the correct way of passing arguments in the array syntax. So please file an upstream issue with Verilator if you would like to see improvements to that, but I'd consider this "works as expected". Either way, there's nothing for us at cocotb to do.

closed 2025-01-29 by imphil