cocotb/cocotb #4399: Replace "units" arguments with "unit"

cmarqu, 2025-01-28:
We should replace the "units" argument in functions like e.g. `get_sim_time` with "unit" in 2.0 and do a deprecation in 1.9.

closed 2025-02-14 by ktbarrett

davekeeshan, 2025-05-11:
Why was this changed? It is less than optimal but it is not entirely wrong wording.  

It does break code, I had to wash through a lot of my code to fix for this, and it is a code change that in incompatible with a previous release.  Code using unit(s) will either work on 1.9 or 2.0, usually there is a deprecation phase for a change like this.

I have changed my code, but I suspect this will be a issue you will be raised a few times when to release 2.0.

This is my hack fix for now to allow a module I have released work on 1.9 and 2.0

```python
        try:
            await Timer(reset_length, unit=self.unit)
        except TypeError:
            await Timer(reset_length, units=self.unit)
```

ktbarrett, 2025-05-12:
We plan on adding warning to a yet unreleased version of 1.9.

PR #4455 (2025-02-14): Rename time 'units' to 'unit'
Closes #4399.

Also added a type alias for time units. This actually caught a runtime bug. The default value for `timeout_time` in `TestFactory.generate_tests` was `"steps"` instead of `"step"`.