cocotb/cocotb #4401: A Release() at the end of a test never occurs

rkryan, 2025-01-29:
In the following example code, the intent is to remove all applied Forces at the end of the test. If Release() is the final operation in a test, the Release never occurs, which leads to the Force() persisting to the next test. If instead the simulator is advanced before finishing the test, such as with the RisingEdge call, the Release does occur.

```
@cocotb.test()
async def test_reg_read(dut):
    await init_model(dut)

    try:
        value = random.getrandbits(64)
        dut.regs.reg_q.value = cocotb.handle.Force(value)
        read_data = await reg_read(dut, 0xabc0)
        assert read_data == value, \
            f"Actual read data 0x{int(read_data):016x} != expected read data 0x{value:016x}"
    finally:
        dut.regs.reg_q.value = cocotb.handle.Release()
        # await cocotb.triggers.RisingEdge(dut.main_clock)
```

This was initially found with a VHDL model with Xcelium.

This does also lead to the tangential general question of Forces persisting between tests if they are not Released. This is the current behavior, but can cause some unexpected behavior if care is not taken.

ktbarrett, 2025-01-29:
There are a couple issues here.

1. The write scheduler is cleaned up as soon as the test ends, not allowing the last set of writes to occur.

These lines should be moved to `_schedule_next_test`.
https://github.com/cocotb/cocotb/blob/09c268647a42087d45f95396381785cdf91253e1/src/cocotb/regression.py#L508-L509

2. If the test fails in some other Task, the `finally` block is never run. This requires implementing `CancelledError`, which should come in with #4280 some time soon.

ktbarrett, 2025-01-31:
This is a little more complicated than I thought. The issue is that `_do_writes` task must survive after the end of the test, or the task and callback will be cancelled, not just the writes. Unfortunately, because the scheduler only works on Tasks, and all Tasks must be killed at the end of the test, that means we would have to introduce a special case into the scheduler to prevent the `_do_writes` Task from being cancelled between tests. 

Seeing as I intended to totally rewrite the scheduler and test shutdown in #4280, I don't see much of a reason to fix this now. It will have to wait until that PR comes in sometime in the next couple weeks ideally.

ktbarrett, 2025-02-22:
The other part of the issue is that these writes *shouldn't be scheduled at all* (#4466), but should be applied immediately. #4479 should solve some of your issues.

ktbarrett, 2025-02-22:
Hooray for you running into an issue that undercuts all of the major issues with cocotb right now that require major surgery to repair!

ktbarrett, 2025-03-13:
With #4503, #4514, and #4524 in, all possible angles of this issue are resolved.

The following should work.

```python
@cocotb.test()
async def test_reg_read(dut):
    await init_model(dut)

    try:
        value = random.getrandbits(64)
        dut.regs.reg_q.value = cocotb.handle.Force(value)
        read_data = await reg_read(dut, 0xabc0)
        assert read_data == value, \
            f"Actual read data 0x{int(read_data):016x} != expected read data 0x{value:016x}"
    finally:
        dut.regs.reg_q.value = cocotb.handle.Release()
```

closed 2025-03-13 by ktbarrett

rkryan, 2025-03-14:
Thank you for the fixes @ktbarrett! I didn't realize what a large thread I ended up pulling on with this issue, but I appreciate the amount of change that had to go in for this.

ktbarrett, 2025-03-14:
It's no problem. They were all on my radar as cleanup I wanted to get to, so this issue gave me motivation to fix it.