cocotb/cocotb #4404: Questa `vhpi_control` applied immeidately

ktbarrett, 2025-01-29:
Calling `vhpi_control(vhpiFinish)` in Questa's VHPI does a clean `exit()` immediately. I don't think this is correct per the spec.

From IEEE 1076-2019 23.5, emphasis mine.
>  If command is vhpiFinish, ***after control returns to the tool from the callback function from
which the vhpi_control function was invoked***, the tool enters the termination phase (see 20.10).

_Originally posted by @ktbarrett in https://github.com/cocotb/cocotb/issues/4392#issuecomment-2620735459_

imphil, 2025-01-29:
Filed [Case 01403593](https://support.sw.siemens.com/de-DE/support-case/details/87f122e0-de72-11ef-ba18-a38bffe2dff1) with Siemens.

imphil, 2025-02-11:
Feedback from Siemens:

I've tested the tool's behavior using a simple test case that utilizes both vhpiFinish and vhpiStop, as follows: 

```c
    vhpi_control(vhpiFinish);
    printf("This should print if vhpifinish works correctly.\n");
 
    //vhpi_control(vhpiStop);
    //printf("This should print if vhpistop works correctly.\n");
```

When using the vhpiFinish, the tool terminates immediately without printing the message. However, when using the vhpiStop, the tool prints the message and then stops the simulation.

Based on my understanding of IEEE 1076-2019 (Section 23.5), after the control returns to the tool, it should enter the termination phase, which appears to align with how Questa behaves.

Could you please clarify what behavior you would expect in the case of Vhpi_finish?  I've attached the test case I used, feel free to modify it to show the tool issue. 

[Note Philipp: Attachment is coming, I can't download it yet.]

ktbarrett, 2025-02-12:
I don't know where the confusion is. 23.5 states:

> after control returns to the tool from the callback function from
which the vhpi_control function was invoked

Which is unambiguous plain English to me, a native English speaker. Exiting before the callback returns is wrong. Questa is also the only simulator supporting VHPI that does this.

The difference between `vhpiFinish and `vhpiStop`, which is not at question here, is whether the simulator should terminate or drop into a state to accept further commands (such as an interactive TCL command line).