cocotb/cocotb #4408: Results hang on last test with GHDL and Cocotb (master branch)

maxime-lr, 2025-01-29:
Got a problem with Cocotb on master branch.
The test runner seems to hang on results with GHDL.

I ran `git bisect` and found that the bug appeared with commit https://github.com/cocotb/cocotb/commit/276e473231beb1cdbfce63d3123da0d5d27d8966

ktbarrett, 2025-01-29:
Can you give your simulator version and a reproducer test. GHDL is working fine in CI.

maxime-lr, 2025-01-29:
> Can you give your simulator version and a reproducer test. GHDL is working fine in CI.

yup sorry, I hit the ENTER key while copying MWE files...
done now!

ktbarrett, 2025-01-29:
There is a process in the GHDL which would continue to run if cocotb stopped without calling `gpi_sim_end` to request simulator shutdown, so that's probably why it's hanging. This seems like a bug, I'll look into it later.

ktbarrett, 2025-01-31:
This is an upstream issue and dupe. I suggest following the tagged issue to get updates when it is fixed upstream and confirmed here.

closed 2025-01-31 by ktbarrett