cocotb/cocotb #4415: Link to examples in quickstart always points to `master`

cmarqu, 2025-02-01:
The link to example code in https://docs.cocotb.org/en/stable/quickstart.html points to `master` but should point to a release tag if not "Latest" docs.

cmarqu, 2025-02-03:
I didn't notice this before, but it works in the `latest` version of the docs (which since recently use sphinx-book-theme, but not sure how this would be relevant), where it will be pointing to e.g. `https://github.com/cocotb/cocotb/blob/d545a5e45053dac67e75d55be049df075e45d72f/examples/doc_examples/quickstart`

The code handling this is
https://github.com/cocotb/cocotb/blob/d545a5e45053dac67e75d55be049df075e45d72f/docs/source/conf.py#L87-L93

rithikasridharbabu-2007, 2026-11-07:
Hi, I’d like to work on this. The cause is in docs/source/conf.py: commit_id comes from git rev-parse HEAD and falls back to master when that fails. I’d read READTHEDOCS_GIT_COMMIT_HASH first (Read the Docs provides it), and also catch OSError so a build without git installed falls back instead of failing. I have it ready on a branch: https://github.com/rithikasridharbabu-2007/cocotb/tree/docs/fix-reposrc-commit-4415. Could I be assigned?

ktbarrett, 2026-11-08:
I'm not certain this issue is relevant anymore. Please **_confirm_** the issue before _solving_ it.

closed 2026-11-08 by ktbarrett