cocotb/cocotb #4434: `get_interpreter_path` fails to get a Python interpreter

matiasilva, 2025-02-04:
Hi,

I'm getting the following output when running a simple test with the Xcelium runner:

```
     -.--ns ERROR    gpi                                ..mbed/gpi_embed.cpp:67   in get_interpreter_path            PYGPI_PYTHON_BIN variable not set. Can't initialize Python interpreter!
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:102  in gpi_print_registered_impl       VPI registered
SVSEED default: 1
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:135  in _embed_init_python              Using Python interpreter at /a/b/c/sim/.venv/bin/python
```

I've tried tracing the `PYGPI_PYTHON_BIN` environment variable to https://github.com/cocotb/cocotb/blob/f3c2b365f8f1dc4acc6f31aa286591cb426e23e9/src/cocotb_tools/runner.py#L190 and I found that it is indeed set, which is how `_embed_init_python` is able to find it, but somehow `get_interpreter_path` fails to? 

Is this expected or is there something wrong in my setup?

Thanks!

ktbarrett, 2025-02-05:
What simulator and simulator version are you using? It seems to both fail and succeed, so I'm guessing what's happening is that cocotb is loaded twice. The first time it fails, the second time it succeeds. This can happen on some simulators, for example, something similar happens on VCS because the cocotb libraries are loaded both during compilation and simulation and it fails first during compilation.

matiasilva, 2025-02-05:
Ah, I see. I'm running Xcelium version 24.09-s001

ktbarrett, 2025-03-25:
Sorry I've been busy gearing up for the 2.0 release. Is this still an issue? Do you think you can provide an example that fails for us? We have Xcelium licenses to test these kinds of things.

matiasilva, 2025-03-27:
I've since integrated cocotb into our own Makefile flow and not seen this again. I was seeing it when I was using the Python runners to run tests. I've been using cocotb master throughout all of this, so I wouldn't be surprised if active work on that fixed things?

closed 2025-03-27 by matiasilva