cocotb/cocotb #4474: Potential live lock scenario with `Lock`

ktbarrett, 2025-02-18:
For example, if a Task acquiring a Lock constantly gives up an acquisition and then attempts to reacquire, which resets it's position in the scheduler queue, it may live lock. Tasks should probably retain their spot in the Lock's queue in case this happens. Essentially the Lock should operate not on a queue of `_Lock`s, but on a queue of Tasks. This is similar to what `cocotb.queue.Queue` does.

Siddharth-singla, 2026-11-11:
I'd like to help with this one, but I want to make sure I understand the behavior you want.

I reproduced the position-reset effect by running cocotb's event loop without a simulator (so it only involves Lock, Event and Task, no HDL). Scenario:

Task H acquires the lock and holds it.
Task R starts lock.acquire() and is first in the queue, then cancels that attempt and calls lock.acquire() again (like a retry loop).
Tasks W1 and W2 then queue up, in that order.
H releases the lock.
Order of acquisition: W1, W2, R. So R lost its place by retrying, as the issue describes.

For comparison, asyncio.Lock gives the same order in the equivalent scenario, so this is the usual behavior of a cancelled and re-requested lock, not an implementation accident.

Questions:

1. Is that scenario the one you meant, i.e. a task that cancels a pending acquire() and then asks again should keep its original place in line?
2. If so, how would the lock recognise "the same task" asking again? The current _Lock object is created per acquire() call, so a queue of Tasks (as you suggested) seems to be the idea. Should a cancelled attempt keep a reservation for that task, and for how long?
3. Would you want a behavior change here, or only a documentation note that cancelling and retrying puts the task at the back of the queue?

What I did NOT test: I did not run this under an HDL simulator, and I did not look at a live lock where time passes (for example with_timeout(lock.acquire(), ...) in a loop), only the ordering above.

ktbarrett, 2026-11-11:
Look at the implementation of `Queue`. We keep a data structure keyed off the current task to look up if the Task still has a position in the queue.

ktbarrett, 2026-11-11:
> 1. Is that scenario the one you meant, i.e. a task that cancels a pending acquire() and then asks again should keep its original place in line?

Yes.

> 2. If so, how would the lock recognise "the same task" asking again? The current _Lock object is created per acquire() call, so a queue of Tasks (as you suggested) seems to be the idea. Should a cancelled attempt keep a reservation for that task, and for how long?

This can be done using `cocotb.task.current_task()`. And yes, cancelling a task should not remove that `_Lock` from the queue. The `_Lock` has a list of tasks to wake up, so when `Lock.release` occurs and a new `_Lock` is acquired, it should first see if the `_Lock` has any active task waiters, if not, the `_Lock` is ignored and it continues to the next one, until a Task actually acquires the Lock.

> 3. Would you want a behavior change here, or only a documentation note that cancelling and retrying puts the task at the back of the queue?

Behavior change, we want to prevent the live lock.