cocotb/cocotb #4500: GHDL `Force` followed immediately by `Release` doesn't release

ktbarrett, 2025-02-22:
Even after #4479, GHDL still has issues with Force and Release, specifically wrt to a Force followed immediately by a Release staying Forced.

From test_force_release
```
@cocotb.test(expect_fail=SIM_NAME.startswith("ghdl"))
async def test_force_followed_by_release_registered(dut):
    """Test if Force followed immediately by Release works on registered signals."""

    cocotb.start_soon(Clock(dut.clk, 10, "ns").start())
    dut.stream_in_data.value = 90

    # Force driven signal then immediately release it.
    dut.stream_out_data_registered.value = Force(5)
    dut.stream_out_data_registered.value = Release()

    # Check if the driven signal is actually released.
    await ClockCycles(dut.clk, 2)
    assert dut.stream_out_data_registered.value == 90
```

ktbarrett, 2025-02-22:
Pinging @tgingold since that worked last time 😄

tgingold, 2025-02-23:
According to the LRM, force has priority over release when executed in the same delta cycle.

Should it be different for VPI ?

ktbarrett, 2025-02-23:
Where do you see that? Maybe I missed it, but I checked section 22.5.

tgingold, 2025-02-23:
See 14.7.3.2: release are processed before the force.

ktbarrett, 2025-02-23:
Wow, that whole cycle is a bit odd. Forcing immediately updates the effective value, but whether the signal is forced or released is decided once at the end of the simulation cycle. I guess that makes sense due to resolving multi-driven nets where one might be Releasing and the other Forcing, but that makes life difficult for us. I guess we will just have to leave this behavior undefined and maybe document that this is a possibility.

closed 2025-02-23 by ktbarrett

ktbarrett, 2025-02-24:
@tgingold While I have you., could you point me to the section that states that released signals should be given the value `U`?

tgingold, 2025-02-24:
There is no such statement.
Testcase ?

ktbarrett, 2025-02-24:
14.7.3.3 Step e.3 states 

> If S is an unconnected signal port of mode in, the effective value of S is given by the default
value associated with S (see 6.4.2.3).

For `std_(u)logic` this is `U`.

Step e.1 states

>  If S is a signal declared by a signal declaration, a signal port of mode out or buffer, or an
unconnected signal port of mode inout, then the effective value of S is the same as the driving
value of S.

An unconnected internal signal would be undriven and default to `U`, correct?

<details><summary>Testcase 1</summary>

```vhdl
-- testcase1.vhd
library ieee;
use ieee.std_logic_1164.all;

entity testcase1 is
    port (
        a_in : in std_logic_vector(7 downto 0);
        a_out : out std_logic_vector(7 downto 0)
    );
end entity testcase1;

architecture behav of testcase1 is
begin
    a_out <= a_in;
end architecture behav;
```

```vhdl
-- testcase2.vhd
library ieee;
use ieee.std_logic_1164.all;

entity testcase2 is
    port (
        a_out : out std_logic_vector(7 downto 0)
    );
end entity testcase2;

architecture behav of testcase2 is
    signal a_in : std_logic_vector(7 downto 0);
begin
    a_out <= a_in;
end architecture behav;
```

```
# testcases.py
import cocotb
from cocotb.handle import Force, Release
from cocotb.triggers import Timer

@cocotb.test
async def test_release_driver(dut):
    dut.a_in.value = Force(1)
    await Timer(1, "step")
    assert dut.a_in.value == 1

    dut.a_in.value = Release()
    await Timer(1, "step")
    cocotb.log.info(f"{dut.a_in.value=}")  # prints the value after Release

@cocotb.test
async def test_release_driven(dut):
    dut.a_in.value = 3

    dut.a_out.value = Force(1)
    await Timer(1, "step")
    assert dut.a_out.value == 1

    dut.a_out.value = Release()
    await Timer(1, "step")
    assert dut.a_out.value == 3
```

```Makefile
# Makefile
SIM := nvc
TOPLEVEL_LANG := vhdl
VHDL_SOURCES := testcase1.vhd testcase2.vhd
COCOTB_TEST_MODULES := testcases

include $(shell cocotb-config --makefiles)/Makefile.sim

testcase1:
	$(MAKE) TOPLEVEL=testcase1

testcase2:
	$(MAKE) TOPLEVEL=testcase2
```
</details>

tgingold, 2025-03-01:
Yes, an unconnected signal would appear as 'U'.