cocotb/cocotb #4506: Graceful method to test for signal presence

markusdd, 2025-02-24:
Hi,

while writing a generic testbench component (a waveform log handler which forwards log messages to string signals viewable in the waveform)
I stumbled across a limitation of the SimHandle API, at least so it seems:

```python
class WaveformLogHandler(logging.Handler):
    """Log Handler which forwards messages to waveform signals."""

    def __init__(
        self,
        dut,
        log_dbg: Optional[SimHandle] = None,
        log_inf: Optional[SimHandle] = None,
        log_chk: Optional[SimHandle] = None,
        log_wrn: Optional[SimHandle] = None,
        log_err: Optional[SimHandle] = None,
        log_ech: Optional[SimHandle] = None,
        log_all: Optional[SimHandle] = None,
    ):
        super().__init__()
        self.dut = dut
        import debugpy
        debugpy.listen(5678)
        debugpy.wait_for_client()
        debugpy.breakpoint()
        logging.getLogger().info("Installing Waveform Logger.")
        a = dut.log_dbg
        try:
            self.log_dbg = log_dbg if isinstance(log_dbg, SimHandleBase) else dut.log_dbg
        except AttributeError:
            self.log_dbg = None
...
```

We have 7 optional signals log_dbg, log_wrn, log_err etc. that if present in the system verilog testbench top should be found by the handler. Then in the emit method of the handler we intercept the logging messages and assign them to the signals.

If all are present this works beautifully.

But if in a given testbench not all of these signals exist we gracefully want to handle this and just not emit to the signal.

In the above example I try this by catching the AttributeError, which works in the sense that the test itself runs PASS.
But our log parsing then gets very upset because for each signal, we get this on the console:

```
INFO     ERROR:          VHPI       NOTOTM                                                                                                                                                                                       
INFO              Cannot iterate on type vhpiInternalRegions using a reference handle of type <unknown>.                                                                                                                         
INFO              /xxxxxxxx/tb.sv, 29:                                                                                                                                                                                                              
INFO     Simulation complete via $finish(1) at time 1 PS + 0                                   
```

Obviously we do not want to filter these messages because in other cases they could represent a real issue.

In a pythonic sense each dictionary has a 'get' method which gracefully returns None without exception.
I have found no such API for cocoTb to test for presence of a signal before trying to use it.

I get the same errors when using the _id("signal_s", extended=False) indirect method. It also thrwos an AttributeError and the above Error is logged, even if caught with try/except.

So long story short: How can we gracefully test for signal presence without causing exceptions?
(I hope this is not an embarrasing oversight while reading the docs xD)

cmarqu, 2025-02-24:
> see if your simulator has a way to turn off VPI error reporting.

Xcelium has `-plinowarn`.

markusdd, 2025-02-24:
`-plinowarn` is already being used and the printouts still happen.

I also checked the xrun.log and the messages indeed are from the simulator.

What really strikes me also is that the messages are from VHPI, not VPI.
The design in question in pure verilog, why would the VHPI feel it needs to respond here?

ktbarrett, 2025-02-24:
> I also checked the xrun.log and the messages indeed are from the simulator.

I'm guessing xrun.log only contains things logged from the simulator and not cocotb? If so there isn't much we can do about that if `-plnowarn` is being passed.

But cocotb has `VHPI_CHECKING` and `VPI_CHECKING` in the C++ code that is enabled by a define and is built by default. We could and probably should, regardless of this issues, get rid of that define and transition these to `LOG_DEBUG`s.

markusdd, 2025-02-24:
xrun.log has not other cocotb logging, correct.

Can you confirm that the cocoTB GPI/VPI code does not contain these error codes (CANTIT, NOTOTM)? I at least could not find them. Also they are capital 6-letter acronyms, which is super typical for Cadence.

But they seem to log differently in VPI, normally warnings are prefixed with W*XXXXXX and errors as E*XXXXXX .
I was not able to find anything about them yet.

As for GPI_EXTRA: yes our setup certainly also loads VHPI, but there are no vhdl files in the design and tb where I see the issue.
As I am just iterating over, or accessing, signals in a systemverilog tb toplevel, I think VHPI really should not be at play here.

I am fully willing to just suppress VPI messages from the simulator, just because if something goes wrong cocoTB will tell us on Python level anyway. But I have not found a useful way yet.

Having the ability to conditionally add TB functionality depending on what the design uses really is key for us, and testing for signal presence is needed for that to make it more user friendly, so a way to get rid of that spam really would be needed.

The reason that the errors are printed is probably because `plinowarn` really just ignores warnings.

The thing I am wondering: Being able to iterate over a HierarchyObject is a nice and common feature of cocoTb, so surely I can not be the first person to have this problem? Also xcelium is a very commonly used simulator, if not THE most common.
Surely someone must have the magic knowledge to work around this?^^

ktbarrett, 2025-02-24:
> As for GPI_EXTRA: yes our setup certainly also loads VHPI, but there are no vhdl files in the design and tb where I see the issue.
> As I am just iterating over, or accessing, signals in a systemverilog tb toplevel, I think VHPI really should not be at play here.

Ahhh, yes. The issue is that since VHPI is loaded after VPI fails to find objects, it will iterate over VHPI to find objects. It does this in case there is a language border that prevents finding an object with one interface that the other interface can handle. There really isn't a way to prevent this without breaking mixed language support considering that the VPI has no way to discover or mark objects that belong to another interface.

The only solution we can offer is to simply not have VHPI loaded so it doesn't try this.

The next is Xcelium actually improving their VHPI.

The least likely is the Verilog standards committee defining the equivalent of `vhpiVerilogModule` and then every simulator implementing that and the VPI equivalent.

> Also xcelium is a very commonly used simulator, if not THE most common.

In total use? Possibly, if not a close second or third. With cocotb? Not really. It's had far too many issues for far too long and will take a certain amount of time for us and them to synergize and people to pick it up. I've had past employers who had many interactions with Synopsys and this is par for the course.

ktbarrett, 2025-02-24:
I think now it's best to get in contact with Cadence. `-plinowarn` should suppress these issues, seeing as the issue is recoverable (just don't return an iterator, or return an empty iterator) and not actually an error.

I will work on fixing the `VPI_CHECKING` and `VHPI_CHECKING` tonight in case their calls to `vpi_chk_error` are what are triggering the issue.

markusdd, 2025-02-24:
much appreciated.
What would be the best way to test this then? I assume current master branch already has 2.0 API changes?

ktbarrett, 2025-02-24:
It should rebase fairly easily on stable. It will probably be a 10 line code change.

markusdd, 2025-02-24:
thanks so much.
will have to check how to pull it into the project setup first thing tomorroww morning, currently we pull from PyPi through a cache, but I'll figure that out.

ktbarrett, 2025-02-25:
https://github.com/cocotb/cocotb/pull/4511 Should be enough for you to test that out.

markusdd, 2025-02-25:
Hi, tested it just now, unfortunately the messages prevail so they are from the simulator and not triggered by CocoTB directly.

Will try some more stuff and come back with it, maybe we can at least get a docs update out of this so people know what to do when they are faced with this.

markusdd, 2025-02-25:
Update:
So there is absolutely nothing to be found in Cadence docs about NOTOTM and CANTIT error codes.
Is it possible that they emits these codes as part of calling vpi_chk_error() and cocoTb is doing the formatting and printing?

I now tried also setting -plinooptwarn in addition to -plinowarn and also -NOWARN NOTOTM and -NOWARN CANTIT and the messages do stay.

ktbarrett, 2025-02-25:
I have an idea related to that. I'll throw something up soon.

markusdd, 2025-02-25:
I will have a look again using your updated PR.

I meanwhile already have a cadence case and they claim these messages have to be thrown according to the IEEE SV standards around PLI/VPI/VHPI.
Even if: would be nice to be able to turn them off^^

I gave them some logs and told them again I just want an off-switch^^

ktbarrett, 2025-02-25:
Thanks! I'm glad they got back so quick. Sucks that's the answer though.

markusdd, 2025-02-25:
yep...I tried your updated PR, no change on my end for that, messages are still visible.

markusdd, 2025-02-25:
As for another idea: If we cannot get these NOTOTM messages to go away when trying to access the handle and catching the exception Python:

Can we do something about the CANTIT situation?

When iterating over `dut` this seems to happen for certain handle types. But truly: I do not even want to 'iterate' over them as well, I just want to know what is present on that level of hierarchy. Unfortunately when triggereing the __iter__ of a RegionObject the 'discover_all' subfunction gets triggered which probably tries to 'brute force' a hierarchy sweep, probably triggering those messages. (and there can be hundreds even for only medium sized testbench toplevels).

So could you think of a method to handle this more 'gently' and not waking up the log demons? (haha)

I think this problem was existing in #4099 as well but with VCS.

Essentially, as it stands now, as soon as you just try to iterate over a Hierarchy or RegionObject you blow up the simulation log^^

markusdd, 2025-02-25:
And for the NOTOTM messages:

```
0.00ns DEBUG    gpi                                Searching for log_dbg                                                                                                                                          
0.00ns DEBUG    gpi                                Unable to find 'tb.log_dbg'                                                              
0.00ns DEBUG    gpi                                Already checked VPI implementation                                                                                                                             
0.00ns DEBUG    gpi                                Checking if log_dbg is native through implementation VHPI                                                                                                      
0.00ns DEBUG    gpi                                VHPI: Unable to query vhpi_handle_by_name tb.log_dbg                                     
0.00ns DEBUG    gpi                                Failed to find a handle named log_dbg via any registered implementation                        
```

you are correct, it cannot find it in VPI context (which does not seem to trigger log errors by the simulator) and then defaults to VHPI.
Unfortunately in the general case we need to assume VHPI is present as we have designs that contain VHDL subblocks. The tb toplevel is always V/SV though.

Is there a way to prevent that if the hierarchy being queried is a verilog block?

ktbarrett, 2025-02-26:
> I just want to know what is present on that level of hierarchy

Unfortunately the only way to know is to iterate. That's the only feature VPI provides. You can either iterate over an object, or you can get a handle by name if you know it.

> Is there a way to prevent that if the hierarchy being queried is a verilog block?

We need to be able to, from a Verilog block, decide to find a sub-object via the VHPI instead, otherwise mixed language simply does not work. And because we have no way of discovering objects from other implementations because VPI has no way of telling us "I found an object, but it belongs to another interface" (`vhpiVerilogModule`), we always have to test every region to see if an object from an external interface is present.

Yeah that means that we are iterating over a Verilog region with VHPI and get errors, but the language designers and simulators force our hands with their lack of consideration. I don't see how this is solvable without language or simulator support.

The VPI does allow for custom `vpiType` response. Perhaps Xcelium can provide a `vpiVHPIObject` if it's so worried about us doing a VHPI iteration of Verilog regions.

And finally, cocotb *definitely* needs to move towards simulator-specific GPI implementation libraries rather than assuming at the top-level we can just switch interfaces without issue. That way we can introduce more simulator-customized mixed language support. #2347, #4496.

markusdd, 2025-02-26:
All makes sense.

What I was thinking about is more a user-induced query.

`_id()` already has the `extended=` option which essentially is there to distinguish between verilog and VHDL to handle the escaping when asking for a signal.

I had some more chat with Cadence today and essentially it's only their VHPI which is misbehaving (probably because there is no VHDL at all in the given example) and ignoring the -plinowarn option.
With VPI we are also hitting a dead end and it does not print such an error.

So it would be nice to be able - from the python space, to limit signal search to a given implementation.
We know our tb top is never, ever going to be VHDL so skipping that search would be a perfectly valid option.

You think something like that would be feasible?

ktbarrett, 2025-02-27:
So you want to, with *every* level of the hierarchy, provide the GPI with the information as to the interfaces to search for, for *all* of your designs, from here forward? And it will still issue errors on levels of hierarchy where mixed language boundaries exist because it iterates over the entire hierarchy and not just the couple elements you care about? All to solve the issue with seeing definitely not intended-to-be-parsed-by-programs log messages?

This is where I turn it back on you and ask you if this is actually a reasonable solution to a real problem.

You can add handlers to Python loggers and have logs written to files that aren't stderr (stderr is for user's eyes, not programs). And those handlers can actually print the whole logger name, function name, file and line number, etc. GPI loggers use Python loggers once cocotb is loaded, so it all "just works" if the GPI has errors to report.

markusdd, 2025-02-27:
No, I think you misread me here^^

I think the iteration problem is not really fixable, because of the reasons you stated, it's just a limitation of the PLI interfaces.

I was just referring to the initial idea, where we ask for ONE specific signal, and we know what area it is in (in our case the very top which is ALWAYS verilog, hence VPI).

In the above example we want to provide the feature to log into the waveforms via the listed signals, but we also aim to provide more 'automatic hookups' (e.g. if there is an APB, autoconnect our cocoTB APB master, and more stuff like that).

The whole idea is to eliminate a lot of boilerplate code by having a 'testbench stub' that is brought in as a one-liner and then hook up to the DUT as needed.

Problem now is: If I do that with 100 optional signals I get spammed with 100 of these VHPI messages although I 100% know that these signals are never, ever going to be in a piece of VHDL code.

Obviously, this is just for the testbench level, in ONE piece of code. This si not about descending into a design that can be anything, as you rightfully stated. That would be unusable madness^^
In normal testcases no one is expected to, nor wants to use cocoTB code that explicitly searches for optional signals via a specific implementation like VPI. The signals used there HAVE to be there, obviously, that is part of the test.

So this is just for generic TB code to reduce boilerplate for our purposes. Essentially the functionality of the _id() function, just extended by something like `impl=` (as optional) would suffice. With the simple consequence that the function at https://github.com/cocotb/cocotb/blob/b4050d5d8406de8c6216b08ed627f108b8fd63e7/src/cocotb/share/lib/gpi/GpiCommon.cpp#L293 would skip the for loop for additional interfaces. (as said, a skipImpl API is even there, it's just not made visible in the Python interface, it's forced NULL).

The log implementation I opened this ticket for works perfectly fine, we just have this annoying spam for our small TBs now that do not use a testbench wrapper and hence are lacking those hookups (and frankly do not need them). This is obviously causing acceptance problems as now you have testcases that say pass but have numoerous ERRORS in the logfiles which we cannot get rid of, as they can't be captured/filtered by the Python log API. They end up in xrun.log and everywhere else.

So that's the long story why I'm after fixing this. The intended feature works, it's just tainted by this BS logging of the simulator.

markusdd, 2025-02-28:
To ease the discussion here, I drafted a PR:

https://github.com/cocotb/cocotb/pull/4517

ktbarrett, 2025-02-28:
After some thought there's definitely a way through here that I think we can both be happy with, but I'm busy with work rn, so I'll look at this over the weekend.

markusdd, 2025-02-28:
appreciate the consideration.

I would also add: I think there is another rationale for having an explicit way to forbid signal search across boundaries:

VHDL and Systemverilog allow signal names that might actually be keywords in the other language. I've seen this cause issues with other commercial verification IP as well so probably having that option is a good idea regardless. It should not hurt anything at least.

If you want to move forward with the PR let me know as certainly I'd need to add some testing for it.

markusdd, 2025-03-04:
as #4517 is merged we can close here.

I think a backport is neither needed nor reasonable, so we'll just enjoy the spam until 2.0 comes out haha.

Thanks for all the nice cooperation on this one.

closed 2025-03-04 by markusdd