cocotb/cocotb #4516: Setting individual bit of LogicArray doesn't work

jrwagz, 2025-02-27:
I've ran into a case on the latest code `v2.0.0.dev0+c91ad618` where trying to set an individual bit of a LogicArray to a particular value doesn't work.

Here is my example to reproduce the problem:

`example_dut.sv`

```verilog
module example_dut #(
    parameter WIDTH = 16
) (
    input logic [WIDTH-1:0] in,
    output logic [$clog2(WIDTH)-1:0] out
);
assign out = in;
endmodule

```

`test_example_dut.py`

```python
"""Cocotb testbench for the example_dut module

This test validates that changing individual bits of a LogicArray works as expected.
"""

import os
from pathlib import Path

import cocotb
from cocotb.triggers import Timer
from cocotb_tools.runner import get_runner

@cocotb.test()
async def example_dut_simple(dut):
    """Test simple example_dut

    Args:
        dut (SimHandleBase): Object representing the simulation handle
    """
    cocotb.log.info("Running example_dut_simple")
    for i in range(16):
        dut["in"].value[i] = 0
        await Timer(1, unit="ns")
        assert dut.out.value[i] == 0, f"Failed to set initial value of bit {i} to 0!"
        await Timer(1, unit="ns")
        dut["in"].value[i] = 1
        await Timer(1, unit="ns")
        assert dut.out.value[i] == 1, f"Failed to change value of bit {i} to 1!"
        await Timer(1, unit="ns")

def test_example_dut_simple():
    """Runner for example_dut_simple"""
    sim = os.getenv("SIM", "icarus")

    runner = get_runner(sim)
    runner.build(
        sources=[Path(__file__).parent / "example_dut.sv"],
        hdl_toplevel="example_dut",
        always=True,
        timescale=("1ns", "1ps"),
    )

    runner.test(
        hdl_toplevel="example_dut",
        test_module="test_example_dut,",
    )

if __name__ == "__main__":
    test_example_dut_simple()

```

Running this will fail with the following output:

```text
$ python3 test_example_dut.py 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:117  in _embed_init_python              Using Python interpreter at /usr/bin/python3
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:102  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 11.0 (stable)
     0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
     0.00ns INFO     cocotb                             Seeding Python random module with 1740700460
     0.00ns INFO     cocotb                             Initialized cocotb v2.0.0.dev0+c91ad618 from /usr/local/lib/python3.10/dist-packages/cocotb
     0.00ns INFO     cocotb                             Running tests
     0.00ns INFO     cocotb.regression                  running test_example_dut.example_dut_simple (1/1)
                                                            Test simple example_dut
     0.00ns INFO     test                               Running example_dut_simple
     1.00ns WARNING  cocotb.regression                  test_example_dut.example_dut_simple failed
                                                        Traceback (most recent call last):
                                                          File "test_example_dut.py", line 25, in example_dut_simple
                                                            assert dut.out.value[i] == 0, f"Failed to set initial value of bit {i} to 0!"
                                                        AssertionError: Failed to set initial value of bit 0 to 0!
                                                        assert Logic('Z') == 0
     1.00ns INFO     cocotb.regression                  *********************************************************************************************
                                                        ** TEST                                 STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        *********************************************************************************************
                                                        ** test_example_dut.example_dut_simple   FAIL           1.00           0.00       1581.56  **
                                                        *********************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                         1.00           0.00        205.16  **
                                                        *********************************************************************************************
```

ktbarrett, 2025-02-28:
> `dut["in"].value[i] = 0`

`dut.handle.value` returns a LogicArray value. You are then setting an index of that temporary value. Then that temporary value is going nowhere.

This is equivalent to
```python
val = dut["in"].value
val[i] = 0
del val
```

closed 2025-02-28 by ktbarrett

jrwagz, 2025-02-28:
@ktbarrett , thanks for the quick response and the details on how to handle this properly!  I can confirm that the changes you suggest do indeed fix my problem.

Thanks for all the great work on this project!