cocotb/cocotb #4518: Consider not deprecating len() for Scalars

markusdd, 2025-02-28:
The new 2.0 API seeks to deprecate the len(function) for scalar signals, as now a split has been made beween these and Array objects.

This makes a `logic [0:0] a;` different from `logic a;`. 

I would argue this makes for a much less intuitive to use API as it becomes assymmetric.

Example:

We have a streaming source TB IP that takes a configurable amount of datalanes which can be the same width or not.

Often times streams can be accompanies by frame or valid signals which just make up one bit.

We use the len() function to get information how wide the datalane signals are to do some width checking and provide additional debug infos. Now, with the new API, if I get a datalane which is just 1 bit, I need to make cumbersome ifinstance checks just to find out if I can call `len` in each and every place.

So I would like to ask to re-dicuss/re-consider this decision? I think a more symmetrical API would be significantly easier to use, there is nothing wrong with just returning '1'.

Or: Is there a better new way to accomplish getting signal width which I have overlooked?

ktbarrett, 2025-02-28:
> Now, with the new API, if I get a datalane which is just 1 bit, I need to make cumbersome ifinstance checks just to find out if I can call `len` in each and every place.

This shouldn't be necessary. Are you saying that single bit logic arrays (`logic [0:0] a`) are getting returned as `LogicObject` instead of `LogicArrayObject` just because they are single bit?

I'll add a test for that to the repo because that's a concerning possibility.

markusdd, 2025-02-28:
No, that was just to make the point that these essentially should feel the same API wise.

But you are correct, testing for that corner case is certainly a good idea.

Still, that wasn't my main point.

let's say I have:

```
logic data_a;
logic [4:0] data_b;
logic [5:0] data_c;
```

and I all connect them to my above mentioned streaming source as datalanes.

In my current 1.9.2 code I can just do:

`dl_widths = [len(dl) for dl in self.datalanes]` to get the width of each datalane.

With the new API I would need to write:

`dl_widths = [len(dl) if not isinstance(dl, LogicObject) else 1 for dl in self.datalanes]`

Arguably, that is not an upgrade for readability and maintainability
I would just propose for __len__ of LogicObject to alwways just `return 1` and be done with it.

I am making this argument also out of the experience for working with cocoTB for 10 years and having had to teach a lot of people how to use it, I immidiately recognize that this would be a stumbling stone which would make people wonder why this doesn't work.
I think API symmetry usually is a good design goal and being able to ask a signal for it's width easily should be universally available.

ktbarrett, 2025-02-28:
opened #4519 for the testing. It shows Icarus returns single bit vectors as `LogicObject` instead of `LogicArrayObject` =/ I'll try to work on a fix for that that, but if they are doing something funny we might just have to always return `LogicArrayObject` for Icarus for compatibility reasons.

markusdd, 2025-02-28:
Haha that wasn't my original intent but glad we found it in the process^^

What's your general take on my reasoning regarding the 'non'-deprecation?

markusdd, 2025-02-28:
provided #4520 , can't only complain, gotta do the work as well 😅

ktbarrett, 2025-02-28:
I just checked and `$bits` takes scalar logic objects and doesn't seem to errors, warnings, or linting issues, so there's some provenance. I suppose it's fine.

PR #4520 (2025-03-02): Undeprecate-len-on-scalars
closes #4518

main rationales are:

- API unsymmetry compared to single-bit vectors
- unnecessary code complication in generic testbench code which now would reuqire isinstance calls to get a simple '1' width
- a lot of existing code breakages - we have over 500 places alone that would now require 'isinstance' checks
- keeping it does not hurt anything, it's a simple return 1