cocotb/cocotb #4521: Update QuickStart Guide files/documentation

GabPGomes, 2025-02-28:
Hi, I'm starting to learn cocotb and had a little trouble when reproducing the QuickStart guide example.

The makefile for the quickstart guide (or the documentation) must be updated:
there are many lines like
           `COCOTB_TOPLEVEL`,  `COCOTB_TEST_MODULES`
while in the docs we have
           `TOPLEVEL`, `MODULE`

ktbarrett, 2025-02-28:
Good catch. If you have time please update them!

GabPGomes, 2025-03-01:
What would be the best way to do it? Should I fork the repo and then create a pull request?

ktbarrett, 2025-03-01:
Github has a good guide on [contributing to an open source project](https://docs.github.com/en/get-started/exploring-projects-on-github/contributing-to-a-project). For cocotb specific information there is some information in [the docs](https://docs.cocotb.org/en/latest/contributing.html). The following page has info on how to set up a development environment. Then, just make your changes and open a PR.

GabPGomes, 2025-03-01:
Thank you for your help!
I've just opened the pull request.

ktbarrett, 2025-03-01:
Like I stated on your PR this is actually a misunderstanding. You are reading the "latest" documentation, which is the documentation for the current master branch. You should be reading the documentation that corresponds to the version you are using (probably cocotb version 1.9, A.K.A. "stable" in the docs).

Luckily Read The Docs has introduced some features recently to make this a bit more obvious. I have removed the "latest" version and added the "master" version, which should be slightly more clear it's meant to follow the master branch on this project. And supposedly we should be able to change the slug from "master" to "development" or something some time soon / with the correct permissions (https://github.com/readthedocs/readthedocs.org/issues/10674#issuecomment-2637251395).

Now I will synergize all the documentation / README.

GabPGomes, 2025-03-01:
That's a good idea. Now it is more clear about which documentation version I should use.

closed 2025-03-01 by GabPGomes

cmarqu, 2025-03-03:
The name change from "latest" to "development" is now done, and the README fix is #4525.