cocotb/cocotb #4530: Tests end with:  ERROR    gpi     VPI: Failed to end simulation

markusdd, 2025-03-04:
Hi guys.

While test driving the newest master branch after all the PRs I noticed that this also happened back when we worked on them, this seems to happen consistently.

All tests seem to end with the line in the title, I assume you already know this?

Tried to find an issue addressing this but did not find one. Is this knows and addressed somewhere already? If yes we can also just close.

ktbarrett, 2025-03-04:
IIRC you are using Xcelium, correct? We have been seeing that as well, it seems to be an upstream issue. https://github.com/cocotb/cocotb/issues/4484.

markusdd, 2025-03-04:
ok I see...more incorrect API implementations by comapnies that charge 20k/y per License xD

could we consider a switch to turn the messages off somehow? I think for development etc. they are fine, but having basically each and every test end with an error will raise eyebrowes^^

ktbarrett, 2025-03-04:
Right... Maybe this stuff should also be included in the LOG_DEBUG level so people only see it when they need to figure out why cocotb isn't working correctly. Should be as simple as making the LOG_WARN into LOG_DEBUG.

markusdd, 2025-03-04:
made a small PR

markusdd, 2025-03-04:
btw it even logs as error right now, not warning.

when you say not working correctly: is there any actual adverse effect from that that has been observed?

ktbarrett, 2025-03-04:
If it works, it works. If it doesn't work, cocotb tries to make it work by using things like `m_removed` to prevent calling up on cancelled callbacks. But it will leak GPI objects out the wazoo. And might just be leaking the simulators VPI objects as well, depending on whether it's actually failing or not.

markusdd, 2025-03-04:
I see, Lovely situation xD

ktbarrett, 2025-03-04:
For example of the worst case scenario, see #4379.

markusdd, 2025-03-04:
ok...that...isn't nice xD

One really should assume they somehow have this figured out by now but hey...constructs that have been intriduced in 2005 still do not work 20 years later so here we are I guess :)

ktbarrett, 2025-03-04:
I'm not against adding some hardcodes in there too if it turns out to be a big problem: out-of-memory or massive slowdowns. If Xcelium is just returning the wrong error code, but *actually* removing the callback handle, we can just delete the GPI callback handle. If they are failing to remove the callback, we can't delete it or the next time it fires we'll get a use-after-free and probably a segfault.

markusdd, 2025-03-09:
resolved by #4531

closed 2025-03-09 by markusdd