cocotb/cocotb #4550: Discussion: bool cast of weak drivers

markusdd, 2025-03-17:
[image]

currently the 2.0 style Logic implementation only allows casts to boolean for 1/0 values.

W/L/H leads to a ValueError. For W this makes sense as it is essentially a 'weak X', but L and H are perfectly valid high and low values, usually encountered in pad logic or other scenarios involving Pulls or Open Drain scenarios (think I2C)

Hence the question: has it been discussed whether these should cast to a valid boolean?

ktbarrett, 2025-03-17:
I don't think people deal with weak values all that often and haven't heard anyone give any opinion on them until now.

What do you think should be done? Should L and H *always* be resolved to 0 and 1 in integer contexts, even when resolve_x is not set (ValueError)? Should there be a mode that resolves weak values first but then otherwise works like ValueError? If a new "resolve weak" mode is introduced, should it be default?

Also up for debate is my recent change here that removes L and H always being resolved. https://github.com/cocotb/cocotb/pull/4512/commits/3df14bfe588a513408238de5e181e5f61573eb51

This affects `__bool__`, `__int__`, `to_unsigned`, and `to_signed` IMO.

ktbarrett, 2025-03-17:
I'm actually starting to like my "resolve weak" mode is default idea. There's weak and strong values in Verilog and we currently don't handle signal strength for Verilog values at all, so we are already implicitly resolving weak values there, so we should parallel that for VHDL.

markusdd, 2025-03-17:
Upfront: I do not have a particulary strong opinion on it at all, I just stumbled over it when working on the deprecations in our codebase rgearding .integer and .to_unsigned(), so I started thinking about it and discussing with a colleague.

That being said:
weak 1s/0s are perfectly valid values, they just differ in behavior when something is driven over them. (e.g. a weak 1 will become a clean 0 when a strong 0 pulls it to ground instead of becoming X)
A 1 on an I2C bus or pullup on a pad is a fully qualified 1 in the digital sense. So thinking about it this way, a weak 1 or 0 should resolve to 1 or zero in a boolean sense, because that is what another real circuit would do as well. I think this - generally - would be expected behavior.

And as you say, in verilog this happens kind of implicitly right now anyway.

So I agree with your take: per default, boolean resolution of valid weak values (so excluding U and W which are weak X) should work and return the respective True or False values.

In rare cases where this is not intended having a switch to change that might be beneficial.

Overall it's true weak logic is used rarely, but certainly not 'never'. Especially Memory models like Flash or i2C sensor models often make use of them. Frankly, certain features like clock stall on i2c cannot even be simulated properly with weak values as everything would become X with strong drivers on all ends. And you also need it for pad pullup/down modes etc.

Long story short: I agree, resolving properly per default makes sense.

PR #4551 (2025-03-18): Improve handling of weak values in Logic/LogicArray
Closes #4550. @markusdd 

I made `bool(Logic)` work like ~~it does in VHDL 2019 where 1 and H are `True` and everything else is `False`~~ `int(Logic)`. I made `int(Logic)` also respect weak values since we are doing that in LogicArray with the `int` cast. And since we unconditionally resolve weak values I moved the L/H resolving up so it could cache the values in LogicArray if the value is only `01LH`.

`LogicArray.__bool__` was moved to DeprecatedWarning because instead of keeping it and changing the behavior, I'd like to remove it. It behav [... cut]
closed 2025-03-18 by ktbarrett