cocotb/cocotb #4562: Support structs on input/outputs

hellow554, 2025-03-19:
I'm using the dff example here.

When I modify it like this:

```verilog
// This file is public domain, it can be freely copied without restrictions.
// SPDX-License-Identifier: CC0-1.0

`timescale 1us/1us

typedef struct packed {
    logic q;
} ot;

module dff (
  input logic clk, d,
  output ot oot
);

always @(posedge clk) begin
  oot.q <= d;
end

endmodule
```

where output is now a packed struct instead of a log.

I would expect that I can write the test case now like this:

```python
@cocotb.test()
async def dff_simple_test(dut):
    # ...
    assert dut.oot.q.value == expected_val
```

but it doesn't work because `oot` is a `Logic`, not a struct with the correct fields.

Am I missing something or is this straight not possible for now?!

hellow554, 2025-03-19:
If I'm using `make SIM=verilator` I get a slight different object, but the same error:

```
Traceback (most recent call last):
  File "/tmp/tmp.Jlqsg1gukb/examples/simple_dff/test_dff.py", line 35, in dff_simple_test
	assert dut.oot.q.value == expected_val, f"output q was incorrect on the {i}th cycle"
AttributeError: 'LogicArrayObject' object has no attribute 'q'
```

hellow554, 2025-03-19:
Oh boy! I went down the rabbit hole looking at gpi and specifically at verilators implementation of VPI and it seems that it isn't implemented for my case.

Let's see if I can bring in some more light

ktbarrett, 2025-03-19:
Indexing into packed structs is not allowed so this will never work. This is not a feature we are going to support.

hellow554, 2025-03-19:
So, there's currently no way for me to say "convert those X bits into this verilog structure"?

Because it somewhat defeats the purpose to implement the struct twice in different languages.

I'll take a look into Packtype but having it nativly would be my preferred solution

ktbarrett, 2025-03-19:
> Would I be able to index the struct, if it is unpacked?

Yes. Packed objects are written in the VPI spec as being non-indexable entities. Unpacked objects are intended to be indexable.

> So, there's currently no way for me to say "convert those X bits into this verilog structure"?

No. It's on my radar to implement, but it's a *long* way off. We need something like packtype for modeling packed values in Python, then we need to expand packed object type introspection in the GPI. Both are non-trivial.

closed 2025-04-26 by github-actions[bot]