cocotb/cocotb #4584: cocotb libraries startup refactor

ktbarrett, 2025-03-26:
## How it works now

When the VPI or VHPI entry point are loaded, a startup and shutdown callback are registered which call `gpi_embed_init` and `gpi_embed_end`. `gpi_embed_init` calls `embed_sim_init` and not much else (besides calling `gpi_embed_end` if there's a failure and arguable that's BAD. You don't know if the user properly initialized, calling finalize may blow up). Then it calls into the GPI's entry point.

The GPI's entry point, configures the logger, loads other GPI implementations via `GPI_EXTRA` and then calls `embed_init_python`.

The "embed" library's `embed_init_python` is a hardcoded reference to load both `libpython` and to the PyGPI's `_embed_init_python` method.

The PyGPI's `_embed_init_python` loads the Python interpreter and returns.

Then in the startup callback, `gpi_embed_init` calls into `_embed_sim_init` which *then* loads into Python, initializes the cocotb module, and starts the regression immediately. This means the initialization functionality is on the stack when the first test starts.

## How it should work

The VPI and VHPI entry points should *not* register startup and shutdown callbacks. These callbacks aren't doing anything for either the GPI implementation or the GPI library that can't be otherwise handled by the implementation or the user.

The GPI implementation entry point should load into the GPI entry point which should load `GPI_EXTRA` (or ideally we only need one impl library), and then `GPI_USERS`. Loading libpython before loading the PyGPI will be implemented as a `GPI_USERS` rather than hardcoding libpython into the GPI. The PyGPI entry point will be another `GPI_USERS`.

The PyGPI entry point will take the argc and argv, load the Python interpreter, then immediately load into Python via `PYGPI_USERS`, initialize the cocotb module and start the regression. The first test will be registered to start when the startup callback fires rather than run immediately. There will need to be new GPI and PyGPI interfaces to support registering on the startup and shutdown callbacks.

marlonjames, 2025-07-23:
In general I like the new approach.

We may run into issues with `_process_packages` and `_setup_root_handle` running before the startup callback, as they interact with the hierarchy. If that happens, I think we can still split the initialization from running the first test, either by registering two startup callbacks, or just returning all the way down through the GPI and then calling the GPI-registered startup callback to run the first test before returning to the simulator.

ktbarrett, 2025-12-31:
So one issue with the split initialization is that we don't have access to the `cocotb.top` during test collection. THis is a pretty massive break that I'd rather not do. So perhaps we still do split initialization, but we need the regression manager to start after start of sim time.