cocotb/cocotb #4585: Refactor GPI/PyGPI/cocotb shutdown

ktbarrett, 2025-03-27:
The GPI, PyGPI, and cocotb's shutdown is rather erratic. We are currently battling issues with Questa's VHPI and NVC and modern Python's mishandling of shutdown causing the CI to fail. This has been a long time coming...

## Premature simulator shutdown

This occurs when the simulator decides it's "done." Likely due to a call to `$finish()` or similar, or because the test has left it with no events to service.

### How it works now

The startup process registers `gpi_embed_end` to be called when a shutdown event occurs. When the test signals the end of simulation the shutdown callback is called. `gpi_embed_end` just sets a flag to prevent multiple shutdown attempts, then runs `embed_sim_event`. This is hardcoded to run PyGPI's `_embed_sim_event`.

`_embed_sim_event` calls into Python. cocotb registers `cocotb._init._sim_event` to be run when this gets called. This callback shuts down the regression by failing the remaining tests with `SimFailure` and then runs cocotb package shutdown callbacks. Among the functions called is `gpi_sim_end` which attempts to call `vpi_finish` or similar and deregister the shutdown callback.

### How it should work

While calling `gpi_embed_end` is *correct*, the cocotb callback calling `gpi_sim_end` is *not* correct. The issue is that this case and the following cases, which need calls to `gpi_sim_end`, all share the same code path.

The GPI implementation entry point should *not* register a shutdown callback to run `gpi_embed_end`, but instead allow cocotb to register a separate function for this case by calling into a PyGPI wrapper around a new `gpi_register_shutdown_callback` function. This callback should *not* call `gpi_sim_end` and should fail any outstanding tests and run the cocotb package shutdown callbacks.

## cocotb regression finishes

If all tests in a regression finish then cocotb asks the simulator to shut down for a graceful exit.

### How it works now

After a cocotb regression finishes it runs the cocotb shutdown callbacks then calls `gpi_sim_end` to request the simulator to shut down.
`gpi_sim_end` calls the GPI implementation's `sim_end` which calls VPI/VHPI functions to request simulator shutdown, but also deregisters the shutdown callback. The shutdown actions must be called immediately because there's no chance for them to be called again.

### How it should work

`gpi_sim_end` should be called to request the simulator to shut down. All `gpi_register_shutdown_callback`s should be run *after the current GPI user returns control to the GPI*. There can be more than one GPI user, and all need to know the simulation is shutting down. It must be after the current user returns in case there are issues with re-entrancy (there are with PyGPI).

## GPI implementation or GPI internal code fails

This is when a GPI implementation or the GPI internal code runs into an unrecoverable error.

### How it works now

When GPI implementation code fails, it immediately calls `gpi_embed_end` to signal to the PyGPI of a failure. `gpi_embed_end` calls `embed_sim_event` in the PyGPI, which calls `_sim_event` in cocotb, which fails the rest of the regression, runs the shutdown callbacks, and requests the simulator end with `gpi_sim_end`.

`gpi_sim_end` calls the GPI implementations `sim_end` which is required to deregister the shutdown callback to prevent user code from running after a failure.

### How it should work

The GPI should call all users shutdown callbacks and call `gpi_sim_end` to request the simulator to shut down. Because the GPI implementation may be in an unknown and invalid state, it should not be re-entered, so shutdown callbacks should be removed.

## GPI user code fails

This is when a `GPI_USERS` runs into an unrecoverable error.

### How it works now

Currently the only way this is handled now is when a `SystemExit` is seen in Python and the PyGPI code ends up translating this into a call to `exit(), which calls the shutdown callback on Questa and no other simulator.

### How it should work

This should request the simulator to end with a call to `gpi_sim_end` and set a flag which causes the shutdown callback in other GPI users to run *after* control is returned to the GPI. The shutdown callback for the current GPI user should *not* be called because it may be in an inconsistent state. But because there is no particular mapping of shutdown callbacks to GPI user, this must be on the user. The GPI will call all shutdown callbacks.

oscargus, 2025-05-21:
As a slightly related note, we use nanobind in APyTypes https://apytypes.github.io/ and had issues when using apytypes in a cocotb testbench. The main issue was https://github.com/apytypes/apytypes/issues/533 but that is resolved and maybe was caused by not using the correct build patch for nanobind.

However, there is still an issue with nanobind reporting leaked objects after a cocotb testbench shutting down (and apytypes being used as part of that testbench). I do not remember all the details, but it basically came down to the order of checking vs garbage collection and that cocotb somehow interfered with the "normal" shutdown sequence.

It is not a big deal, we just turned off those warnings, but thought it was worth mentioning. I'd be happy to try PRs for this issue when available to see if it also solves this (and at the same time try to recall what was actually causing it).

ktbarrett, 2025-05-21:
At some point in the future I expect the PyGPI to be reimplemented using either pybind11 or nanobind (not sure how much support nanobind has for embedding Python). I wouldn't doubt there were leaks. And depending upon what version you are working with there have definitely been leaks fixed in the 2.0 development cycle.

Finally we are tracking some issues at my employer wrt entering cocotb from multiple threads and one of the things that fell out was that we aren't releasing the GIL after initializing cocotb and that finalization must occur from the same thread that initialized it. That *may* have something to do with the warnings you are seeing.