cocotb/cocotb #4615: how to print log in runner function

wieniawski-lai, 2025-04-16:
hi,

I am studying cocotb, by the official `adder` example.

I have successfully print log using cocotb.log.info('something') to print log in testcase function `async def adder_basic_test(dut)`

However, I wanna add some print information in `def test_adder_runner()`function:
```python
if hdl_toplevel_lang == "verilog":
        verilog_sources = [proj_path / "hdl" / "adder.sv"]
else:
        vhdl_sources = [proj_path / "hdl" / "adder.vhdl"]
cocotb.log.setLevel(cocotb.logging.DEBUG)
cocotb.log.info(verilog_sources)
```

However, this log.info is not printed out in shell, while execute make.

Does anyone knows how to print log in `runner`(python entrance) function?

thx

wieniawski-lai, 2025-04-17:
> `cocotb.log` is meant to be used within the simulator by the tests ans test modukes. It is not configured the same way here as it is in the tests. I'm surprised it isn't blowing up. Generally, nothing in the cocotb namespace should be touched except by test modules.
> 
> You should make your own logger or use `print`. Additionally, if you are using a Logger, Python defaults all loggers to WARNING level, so info logs won't be seen. You will have to set the level using `setLevel`. And if you are running the build step with `pytest`, `pytest` captures stdout by default, so you won't see any output unless you add the `-s` option to the call.

thx a lot.

I use `print` in tests, like this
```python
if hdl_toplevel_lang == "verilog":
        verilog_sources = [proj_path / "hdl" / "adder.sv"]
else:
        vhdl_sources = [proj_path / "hdl" / "adder.vhdl"]
print(verilog_sources)
```

Only I am using `python test_adder.py` to execute the python file, the print is successfully printed.
However, if I was using `make` to execute, the python's print is still un-seen.

ktbarrett, 2025-04-17:
If you are running with Make, the Python file is never run, so of course you won't see any prints from it. You only need to do one of the Python runner or Makefile, not both.