cocotb/cocotb #4618: VCS python interpreter version mismatch when -load libcocotbvpi_vcs.so

paul-demo, 2025-04-17:
I have cocotb working with VCS 2021 but with 2023 and some new wrapper python, I'm finding that there is some issue with loading the cocotb .so file using the `-load` flag. Specifically, it looks like a python interpreter mismatch, almost as if whatever is loading the .so file is using python 3.6 or 3.7 rather than 3.9 that the .so is created with. This is because the place it fails is somewhere in _py_abc.py which uses the new-to-Python3.8 feature of "positional only parameters" which is this forward-slash syntax

I've experimented with various PYTHONHOME/PYTHONPATH/LD_LIBRARY_LOC settings, but still just get a crash and stack trace due to these issues. Does anyone know if there is a way to force a particular python interpreter to be used when VCS is installed? Is there something else to try when loading the Cocotb .so file?

```python
    def __new__(mcls, name, bases, namespace, /, **kwargs):
                                              ^
SyntaxError: invalid syntax
```

ktbarrett, 2025-04-17:
In 2.0 there's `PYGPI_PYTHON_BIN`, which is used to source the correct Python environment. On 1.9, you can set `VIRTUAL_ENV`, but I'm not sure how that will work with other Python tools as you're basically lying and saying you're running from a virtual environment.

The reason setting `PYTHONHOME` isn't working is likely due to VCS overwriting it so I can call its own Python for various reasons.

paul-demo, 2025-04-17:
We are running form a virtual environment, so maybe I'll give `VIRTUAL_ENV` a shot.

paul-demo, 2025-04-18:
I guess you meant use a virtual environment set to match the version of python the VCS uses; that's probably not going to be feasible for us to roll back to an unsupported python release from before 2019.

My other thought/hope is that if I upgrade to VCS 2024 maybe they have started to use a more modern python release internal to VCS and the issue will just go away.

paul-demo, 2025-04-18:
It seems the issue has gone away with VCS 2024.09, so I'm just going to do that rather than debug VCS 2023.12.

closed 2025-04-18 by paul-demo

ktbarrett, 2025-04-18:
You should not need to use the Python that ships with VCS. We don't in CI and it works fine.

Glad 2024 has the solution. That's actually the first release we have access to.