cocotb/cocotb #4625: TestFactory: It is running the generated tests on the existing dut, Not actual POR

De1shMakwana, 2025-04-21:
Hi,

When I am generating the tests using the TestFactory in the following manner - 
```python
tf = TestFactory(test_function=test_TestFactory)
tf.add_option(name='clk1_period', optionlist=["40n", "100n"])
tf.add_option(name='clk2_period', optionlist=["65n", "68n"])
tf.generate_tests()
```

It is running the test as fresh test, It is continuing the previous test where some registers are keeping their values.
Can anyone suggest how can I run them exclusively?

Thank you,
Devansh.

ktbarrett, 2025-04-21:
> It is not running the test as fresh test, It is continuing the previous test where some registers are keeping their values.

This is how all tests in cocotb run using the RegressionManager.

> Can anyone suggest how can I run them exclusively?

Right now you can use `TESTCASE` and `MODULE` environment variables to select a test, and then run each test you've built.

It would look something like this for your example.
```make
MODULE := my_test_module
all:
    $(MAKE) sim TESTCASE=test_TestFactory_001
    $(MAKE) sim TESTCASE=test_TestFactory_002
    $(MAKE) sim TESTCASE=test_TestFactory_003
    $(MAKE) sim TESTCASE=test_TestFactory_004

include $(shell cocotb-config --makefiles)/Makefile.sim
```

De1shMakwana, 2025-04-22:
It worked,
Thanks!!

closed 2025-04-22 by De1shMakwana