cocotb/cocotb #4631: Support gui for open-source simulators

oscargus, 2025-04-24:
It would be nice to support automatically opening a waveform viewer after simulating with Verilator, NVC, and GHDL (and Icarus Verilog*?).

A preliminary design idea is to add a method to the `Runner` class that returns a filename/Path of the waveform file or None (the Runner base class returns None). Then, after running the simulation one can check if gui and the path is not None and in that case open Surfer or GTKWave, in some order if both are available on the system.

* Not sure if Icarus supports other things than shm and trn or which viewers support those...

MonsieurFantome, 2025-04-30:
Hi, 

The trick is that there are many simulators, that generates many formats and can be opened by many waveform viewers.
And many will want something slightly different than the "standard" cocotb implementation, that's why I tend to gravitate toward custom flow rather than using cocotb's runner.

But I've written the following code to automatically open waves after running a sim.

```python
# Key to select the current waveviewer to open
WAVEVIEWER = "surfer"

# Path dictionnary
WAVEVIEWER_PATH = {
  "gtkwave" : Path("C:/") / "msys64" / "mingw64" / "bin" / "gtkwave.exe",
  "surfer"  : Path.home() / "Applications" / "surfer.exe",    # https://surfer-project.org/
}

# Options
WAVES         = True                    # Record signal traces.
GUI           = True                    # Run with simulator GUI. With GHDL, call gtkwave after running

# Save waves argument
if (WAVES or GUI):
    PLUS_ARGS.append(f"--wave=test_" + f"{test:03d}" + ".ghw")

# Build and run the testbench
cocotb.runner.build(...)
cocotb.runner.test(...)

# Open waves
if (GUI):
    ghw_path = RESULT_PATH / ("test_" + f"{test:03d}") / ("test_" + f"{test:03d}" + ".ghw")
    print(f"INFO: Opening wave viewer : {str(WAVEVIEWER_PATH[WAVEVIEWER])} {ghw_path}")
    subprocess.Popen([str(WAVEVIEWER_PATH[WAVEVIEWER]), ghw_path])
```

oscargus, 2025-04-30:
Yeah, the question is really how much one must be able to reconfigure this. See #4635 for the tentative implementation. It supports setting viewer (defaults to surfer and gtkwave, if any is installed) and handles the simulators that currently does not have a gui flag and produces a format readable by any of those (vcd, fst, ghw). It is not possible to select format though, but it returns that same as the `-waves` flag.

Your script works with GHDL(?), but not any other simulator, so I think the current solution can replace that (and support other simulators). You will have to replace `WAVEVIEWER` with the environment variable `COCOTB_WAVEFORM_VIEWER` if you want to specify viewer, otherwise it will select surfer (if it is in your system path, if not you will have to set the path in `COCOTB_WAVEFORM_VIEWER`).

For me, coming from VUnit, being able to specify these things from the command line in one go, is really a useful feature.

MonsieurFantome, 2025-04-30:
Yeah I just saw your PR and your implementation is very nice compared to my quick and dirty for GHDL.
It will integrates nicely ! Ty

ktbarrett, 2025-04-30:
@oscargus I would not consider the cocotb Makefiles or Python Runner to be equivalent to VUnit, nor is there any interest in doing so. It's designed to be sufficient for internal regression needs (and it doesn't do that very well right now IMO) and sufficient for simple projects.

It mostly exists because there wasn't a good general purpose project automation tool around when cocotb started that could integrate cocotb. Not because cocotb needs to or wants to reinvent the wheel. And the Python runner exists to replace the Makefiles so that we can support Windows without requiring users to install a Unix subsystem (they all have some serious issues we are tired of dealing with). Most serious users develop custom flows or integrate them into their existing one.

That being said, making what already exists better supported and nicer to use is always encouraged. I appreciate the PR.

oscargus, 2025-04-30:
Yeah, I'm not trying to turn cocotb into VUnit, just using it for inspiration on how to solve things that I find a bit tedious in cocotb, like not being able to start the waveform viewer as part of starting the simulation. Also, with my interest in waveform viewers, those aspects will clearly be in focus.

PR #4635 (2025-05-04): Add gui option for GHDL, NVC, Verilator, Icarus and Dsim

Builds on #4630

Closes #4631

Opening to get feedback on design (third commit, until #4630 is merged) and if it may be possible to test it.

- [x] Add news fragment
- [x] Add documentation about the environment variable

Edit: this now allows passing WAVES and GUI using the corresponding environment variables. If this is a good idea (I think it is, they depend on the runs, not the tests), I should also:

- [x] Add documentation about this.