cocotb/cocotb #4632: Upcoming Verilator change affecting waves

oscargus, 2025-04-24:
https://github.com/verilator/verilator/blob/6ac2aa2d99ef1f532232f42d792229f5e9403b4f/Changes?plain=1#L23

Not sure if it will be a clean breaking change or if there is any transition path with a deprecation period.

(Does cocotb check simulator versions somewhere so that it is easy to know which version is used and therefore adapt the commands? If not, maybe the runners should support that?)

ktbarrett, 2025-04-24:
We don't use that. We build our own main loop so `--trace` is our own. Maybe we could/should be using `--binary`? I'm not sure what features it has wrt VPI support.

oscargus, 2025-04-24:
OK! Good!

These were the lines that confused me:
https://github.com/cocotb/cocotb/blob/3ae5690dfa2bc82df772326ee4b1a664c64d92ab/src/cocotb_tools/runner.py#L1313
https://github.com/cocotb/cocotb/blob/3ae5690dfa2bc82df772326ee4b1a664c64d92ab/src/cocotb_tools/runner.py#L1345

But I have no insight into the details ((not even using Verilator, just reading source code...)

Closing then, but feel free to reopen if the `--binary` stuff should be discussed.

closed 2025-04-24 by oscargus