cocotb/cocotb #4659: pyuvm ecosystem tests do not use cocotb 2.0

imphil, 2025-05-07:
Ecosystem tests such as https://github.com/cocotb/cocotb/actions/runs/14435312465/job/40475379567 are supposed to use the master version of cocotb against a released version of a tool. But since pyuvm specifies `cocotb < 2.0` it discards the master version of cocotb and uses the 1.9.2 version instead. That's not that the test is expected to do.

oscargus, 2025-05-08:
Same thing with `cocotb-coverage`:

```
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.2 from /opt/hostedtoolcache/Python/3.11.12/x64/lib/python3.11/site-packages/cocotb
```

Both are fixed in #4668, but exposes errors.

ktbarrett, 2025-05-09:
Yeah that's expected. There's already issues open in both about 2.0 compatability.

PR #4668 (2025-05-20): Install development version of cocotb after ecosystem test package

Closes #4659

(Running as part of the PR to see how it works. Will revert that part once it's been approved.)

I *think* the install works in the expected way now, but errors are triggered using cocotb 2.0.
closed 2025-05-20 by ktbarrett