cocotb/cocotb #4660: Documentation build issues

oscargus, 2025-05-07:
In #4640 some of the documentation build issues are fixed. There are still some to sort out though (that can be fixed). To see the latest build logs: https://app.readthedocs.org/projects/cocotb/builds/

- [ ] Lots of unsupported tags for Doxygen `warning: ignoring unsupported tag`. Most likely a consequence of Doxygen development removing those tags.
- [ ] Lots of `.../docs/source/building.rst:: WARNING: 4 column based index found. It might be a bug of extensions you use: [('pair', 'GUI; Make Variable', 'var-GUI', '')]`. Some extension is using an old format of the index. Would be good to figure out which and consider if it is required.
- [ ] It seems like `setuptools` does not have an API reference documentation, so no links are working.

cmarqu, 2025-05-07:
For the "4 column" issue, see #3314, #2253 etc.

oscargus, 2025-05-07:
It seems like the vendored version of `sphinxcontrib-domaintools` is not installed anymore:

```
Collecting sphinxcontrib-domaintools>=0.1 (from sphinxcontrib-makedomain->-r docs/requirements.txt (line 8))
  Downloading sphinxcontrib_domaintools-0.3-py3-none-any.whl.metadata (814 bytes)
```

oscargus, 2025-05-07:
Closing this and moving the items to #4560

closed 2025-05-07 by oscargus