cocotb/cocotb #4662: Update build mechanism of vendored `domaintools`

oscargus, 2025-05-07:
After #4661 when this was reinstated:
```
  DEPRECATION: Building 'sphinxcontrib-domaintools' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'sphinxcontrib-domaintools'. Discussion can be found at https://github.com/pypa/pip/issues/6334
```

Current pip version is 25.1, so some time, but not a lot.

ktbarrett, 2025-05-07:
I have a feeling we should just move this package repo and let that package manager handle updating it and whatnot. I recently had to do that with [a different sphinx extension](https://github.com/ktbarrett/prettyspecialmethods).

ktbarrett, 2025-05-08:
I forked it and updated it with the new build method. https://github.com/ktbarrett/domaintools

ktbarrett, 2025-05-11:
Closed by #4661.

closed 2025-05-11 by ktbarrett