cocotb/cocotb #4681: Do not run eco-system tests nightly for forks

oscargus, 2025-05-16:
It is a bit annoying to get a mail every day that the eco-system tests fail. It is also a waste of compute that everyone that forks cocotb should run this test.

I think something like
```
if: github.repository_owner == 'cocotb'
```
at a suitable location in https://github.com/cocotb/cocotb/blob/master/.github/workflows/ecosystem-compat.yml should do the trick. For matplotlib, the line is in the script, so the job is executed, just that it doesn't do anything unless it is run from the main repo. 

Maybe there are other jobs that also should have something similar?

ktbarrett, 2025-05-16:
If you don't want Github Actions to run at all, you can disable Github Actions in your fork by looking under settings. I was under the impression that forks disabled Actions by default. We did not implement them for other people to use and recommend not to try to unless you have a good reason.

But assuming that you want to run them you will need to disable all tests that require licenses in CI, extended, and experimental tests. The upload of the performance benchmark tests would need to be disabled. Then the question is whether someone who wants to run CI in their fork wants the extended, experimental, and ecosystem churn tests or not. Maybe they do, maybe they don't.

My take is a lot of this is out of scope. It requires either guessing or prescribing a solution. Maybe disabling everything that isn't the CI test of FOSS simulators is a reasonable universal solution. But if a users cares beyond that, they will need to do work to make it work for them.

oscargus, 2025-05-18:
I'm not sure if they are enabled by default or I did something to have the enabled by default (at least I did not enable them for cocotb explicitly). Yes, one can disable them, but I think it makes sense to behave in a rather reasonable way by default. So I think we agree.

(I've noted this earlier in other packages, but never really bothered to do something about it. But with the eco-system test running nightly and sending an email every 24 hours, I thought I'd try to do something about it... Probably lots of other packages having the same issue. One can wonder how much energy is saved by disabling those tests for all forks of all packages running without people even knowing it...)

closed 2025-05-19 by ktbarrett