cocotb/cocotb #4702: [NVC] Use compact messages to prevent utf-8 errors

AnthoJack, 2025-05-27:
Hi,
I've recently come up with a problem where cocotb failed due to a UnicodeDecodeError after seemingly receiving some non_unicode bytes from NVC
```
<Task finished name='Task-15' coro=<Simulator._log_pipe() done, defined at <repo>/venv/lib/python3.12/site-packages/cocotb_test/simulator.py:254> exception=UnicodeDecodeError('utf-8', b'   Process :reaccess_top:\xa9\xdf\xf0\xe3:ascon_top_inst:ascon_cryptocore_int:_p12 at <repo>/dev/src/CryptoCore_Ascon.vhd:366\n', 25, 26, 'invalid start byte')>
Traceback (most recent call last):
  File "<repo>/venv/lib/python3.12/site-packages/cocotb_test/simulator.py", line 267, in _log_pipe
    self.logger.log(level, line.decode("utf-8").rstrip())
                           ^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa9 in position 25: invalid start byte
```
I didn't manage to find exactely why NVC was giving me this error but after a look at NVC's documentation, I did manage to fix it using the `--messages=compact` option that specifically makes output ["easily parsed by tools"](https://www.nickg.me.uk/nvc/manual.html#messages). It apparently delivers since adding it to the global options of NVC ("extra_args" in Simulator constructor) prevents the error from showing up

This seems like something to either hard-fix in the NVC simulator class at cmd construction or at least warn users about in the documentation chapter about NVC

I'd be happy to submit either of those (or even both) if someone finds any value in it

AnthoJack, 2025-05-27:
OK, well for now maybe this should still be mentioned somewhere in the doc to help?

ktbarrett, 2025-05-27:
IEEE 1076-2019 5.2.2.2 states that VHDL `CHARACTER`s are encoded using ISO 8859-1. So it's reasonable they are using that internally, but it sounds like a bug they are not handling encoding conversions to the users locale when displaying diagnostics despite trying to. `locale`s and encoding conversions in C are a mess so I'm not surprised. I would suggest that they use something like libICU for string encoding conversions.

AnthoJack, 2025-06-01:
> Is your subhierarchy in `Process :reaccess_top:\xa9\xdf\xf0\xe3:ascon_top_inst:ascon_cryptocore_int:_p12 at <repo>/dev/src/CryptoCore_Ascon.vhd:366` using something special, like an extended identifier?

I don't think it's anything special but ascon_top_inst is inside a if-generate statement named "Label_ascon", funny that this name doesn't appear here. I've tried accessing ascon_top_inst using cocotb and failed so it seems like there might be a problem on my part here but couldn't manage to find out what yet

> IEEE 1076-2019 5.2.2.2 states that VHDL CHARACTERs are encoded using ISO 8859-1. So it's reasonable they are using that internally, but it sounds like a bug they are not handling encoding conversions to the users locale when displaying diagnostics despite trying to

I just realised that my issue comes from code that's found in [cocotb_test](https://github.com/themperek/cocotb-test) so maybe posting an issue on its repo would be more coherent but if anyone finds out why my if-generate label gets turned into weird bytes, I'd be very grateful

ktbarrett, 2025-06-01:
This is very likely an NVC issue, please open an issue in that repo.

closed 2025-06-02 by AnthoJack