cocotb/cocotb #4708: Python 3.14 release package regression failing

ktbarrett, 2025-05-30:
Pinging @oscargus. This has been failing since 3.14 support was added. If this can't resolved, 3.14 should be removed from the upcoming 2.0 release. Or we need some way to turn off doing release builds for unreleased versions of Python. 

https://github.com/cocotb/cocotb/actions/runs/15335716488/job/43152913734

cmarqu, 2025-05-30:
Probably cibuildwheel needs to be updated to a newer version?: https://github.com/cocotb/cocotb/blob/fef6cb0e0abf17efc7296c21299d3b4e14e6b801/noxfile.py#L35

Actually, a 2.x release that supports 3.14 appears to be still be pending: https://github.com/pypa/cibuildwheel/issues/2047#issuecomment-2862585414
but there is a 3.0 beta that has support: https://github.com/pypa/cibuildwheel/releases

oscargus, 2025-05-30:
I realized that yesterday. 

It is possible to activate 3.14, but I have not figured out how to do it during the tests, but not during the actual upload build...

(What is strange is that it worked in the PR...)

ktbarrett, 2025-05-30:
> (What is strange is that it worked in the PR...)

The PR tests runs against the source directly. The release test is specifically designed to test against the built wheel to make sure something isn't missing from a releasable build.

> Maybe the easiest thing is to add a separate job for 3.14 that doesn't rely on the nox-stuff and a pre-built wheel? Or just skip it for now...

Maybe it's possible to add a flag to prevent doing a release build and test of certain targets.

oscargus, 2025-05-31:
> The PR tests runs against the source directly. The release test is specifically designed to test against the built wheel to make sure something isn't missing from a releasable build.

Ahh, that makes sense!

Will look into it!

PR #4711 (2025-06-25): Fix CI issues

Closes #4708 
Closes #4709 

Move 3.14-test to experimental.

I realised that we can just run two tests in the same file, so no need to duplicate all the skip-stuff. Also this gives a better dependency chain for the deployment.

Added `cancel-in-progress: true` to stop old test runs after pushing to the same branch again (quite useful for me at least, who tend to push before I think...)

Also added so that all the changed workflows executes on pull-requests. Will remove that once it is confirmed that it works by someone with correct bits.

I do not think the failing tests are ca [... cut]
closed 2025-06-25 by ktbarrett