cocotb/cocotb #4814: Print warnings when unsupported options are used in Python Runner

cmarqu, 2025-07-18:
While reading the Python Runner code to answer #4813, I noticed that we silently ignore the `waves` argument for `build` in Xcelium. 
This could be reported as a warning and is also missing in the docstring at https://github.com/cocotb/cocotb/blob/066639ec8d800942fa3a057de896dd7ca944fdde/src/cocotb_tools/runner.py#L1444

This is also a problem with other options of other simulators.
Tests for this should be added.

ktbarrett, 2025-07-19:
Also we could do like we do in Verilator and carry the arg from the `build` through the `test` command.

cmarqu, 2025-07-27:
> Also we could do like we do in Verilator and carry the arg from the `build` through the `test` command.

Looking through the code, I think I don't understand what you mean here.

PR #4826 (2025-07-28): Improvements around runner
* Adds support for timescale for Xcelium in runner
* Raise exception when waves is set in build step for Xcelium (closes #4814)
* Improve docstrings and Simulator Support section for all simulators
closed 2025-07-28 by cmarqu