cocotb/cocotb #4839: Version 1.9.2 fails to work with Verilator when --trace option is used

mornthx, 2025-08-01:
Using cocotb 1.9.2, Verilator 5.039 devel and python 3.10.
report error when _--trace_  option is used as below:

`Vtop.cpp:125: Turning on wave traces requires Verilated::traceEverOn(true) call before time 0.`

Similar issue as reported in #4141, which is said to be fixed in 1.9.1.

I wonder if it's relevant to the verilator change mentioned in #4632, so I tried --trace, --trace-vcd, --trace-structs and their combination, didn't work.

It's confusing me since Verilated::traceEverOn(true) is sourced in verilator.cpp as below:
`#if VM_TRACE
    Verilated::traceEverOn(true);`

and VM_TRACE is correctly set:
`ccache g++ -Os  -I.  -MMD -I/usr/local/shuai/verilator/include -I/usr/local/share/verilator/include/vltstd -DVERILATOR=1 -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -DVM_TRACE_SAIF=0 `

mornthx, 2025-08-02:
For the first question, I'm sure I do `make clean` every time before `make`. 

For Vtop.cpp , this file seems like a generated source code, so its content is likely to vary.

You might check the file i upload, the `trace configuration` is only generated in Vtop_trace_enabled.cpp, which has 153 lines, 51 more lines than the Vtop_trace_disabled.cpp .

[vtop_example.tar.gz](https://github.com/user-attachments/files/21557058/vtop_example.tar.gz)

ktbarrett, 2025-08-02:
Oh Vtop, I was thinking verilator.cpp. 

I'm guessing this has to do with running callbacks before that call. This is fixed upstream. Have you tried the beta release that has this fix? `pip install cocotb --upgrade --pre`?