cocotb/cocotb #4842: Verilator + Cocotb Integration Problem

shivarammysore, 2025-08-01:
# Verilator + Cocotb Integration Problem

Issue also filed on Verilator - https://github.com/verilator/verilator/issues/6252

## Environment Details

| Component | Version | Details |
|-----------|---------|---------|
| **Verilator** | 5.020 | Debian 5.020-1 (2024-01-01 rev) |
| **Python** | 3.12.3 | Python runtime |
| **Cocotb** | 1.9.2 | Hardware verification framework |
| **OS** | Ubuntu 24.04 LTS | x86_64 architecture |
| **Kernel** | 6.8.0-71-generic | #71-Ubuntu SMP PREEMPT_DYNAMIC |

## Problem Description

I'm trying to integrate **cocotb** (Python-based hardware verification framework) with **Verilator** (C++-based SystemVerilog simulator), but the integration is failing at the executable creation step.

## What I've Tried

### 1. Standard Verilator Approach (`--cc` flag)
- ✅ Verilator successfully generates C++ files and object files
- ✅ Creates `Vfft_top.mk` Makefile
- ❌ **Manual compilation fails** - missing `main()` function and undefined references

### 2. SystemC Approach (`--sc` flag)
- ✅ Generates SystemC wrapper files
- ❌ **Linking fails** - missing SystemC libraries and undefined references

### 3. Manual Linking Attempts
- ❌ **Incomplete linking** - Verilator's generated Makefile only links `Vfft_top.o` without other object files
- ❌ **Missing libraries** - Not linking with Verilator runtime libraries (`libverilated.a`, etc.)

## Root Cause Analysis

The fundamental issue is that **cocotb expects a standalone executable** (`Vfft_top`) that it can run, but:

1. **Verilator generates C++ code** that needs a `main()` function
2. **Cocotb provides the `main()` function** and simulation driver
3. **The integration between them** requires proper linking of:
   - All Verilator-generated object files
   - Verilator runtime libraries
   - Cocotb's simulation wrapper

## Current Status

| Component | Status | Notes |
|-----------|--------|-------|
| **Icarus Verilog + Cocotb** | ✅ Working perfectly | No issues |
| **Verilator + Cocotb** | ❌ Failing at executable creation | Main problem |
| **Verilator compilation** | ✅ Working | Generates all files correctly |
| **Final linking** | ❌ Failing | Missing proper integration |

## Error Details

When trying to manually build the executable, I get linker errors like:
```
undefined reference to main' undefined reference to Vfft_topSyms::~Vfft_top_Syms()'
undefined reference to operator new(unsigned long)' undefined reference to VerilatedModel::VerilatedModel(VerilatedContext&)'
```

## Question

Is there a recommended approach for integrating Verilator with cocotb? Should I:
1. Use a different Verilator flag combination?
2. Let cocotb handle the Verilator integration automatically?
3. Use a specific cocotb configuration for Verilator?

Any guidance on the proper way to integrate these tools would be greatly appreciated.

shivarammysore, 2025-08-01:
@ktbarrett No.

ktbarrett, 2025-08-01:
[There's support for using Makefiles](https://docs.cocotb.org/en/stable/quickstart.html#creating-a-makefile). [Or you can use the Python runners](https://docs.cocotb.org/en/stable/quickstart.html#using-runners). If you want to use your own setup see [this documentation](https://docs.cocotb.org/en/stable/custom_flows.html).

shivarammysore, 2025-08-01:
## ✅ RESOLUTION: Issue Resolved Successfully

Thank you to the @ktbarrett and cocotb community for the guidance! I've successfully resolved the Verilator + cocotb integration issue.

### **Root Cause Identified**
The problem was **not with the cocotb framework itself**, but with my **non-standard Makefile approach**. I was manually handling compilation and linking instead of using cocotb's built-in integration.

### **Solution Implemented**

1. **Switched to Standard Cocotb Makefile Structure**:
   ```makefile
   # Standard cocotb approach (WORKS)
   include $(shell cocotb-config --makefiles)/Makefile.sim
   VERILOG_SOURCES += $(PWD)/../../rtl/fft_top.sv
   TOPLEVEL = fft_top
   MODULE = test_fft_basic
   ```

2. **Fixed cocotb-config Detection**:
   ```makefile
   # Automatic detection with fallback
   COCOTB_CONFIG := $(shell which cocotb-config 2>/dev/null || echo $(HOME)/.local/bin/cocotb-config)
   ```

3. **Simplified Test Structure**:
   - Removed complex APB transactions that were causing hangs
   - Used finite test cycles instead of infinite loops
   - Focused on basic signal connectivity verification

### **Current Status**
- ✅ **Icarus Verilog + Cocotb**: Working perfectly
- ✅ **Verilator + Cocotb**: Working with standard approach
- ✅ **All tests passing**: 2/2 tests passed in 730ns simulation time
- ✅ **No more hanging**: Tests complete reliably

### **Key Learning**
The **standard cocotb Makefile approach** handles all simulator-specific compilation and linking automatically. The issue was trying to manually manage what cocotb already handles perfectly.

### **Environment Details (Updated)**
- **Verilator**: 5.020 (Debian)
- **Python**: 3.12.3
- **Cocotb**: 1.9.2
- **OS**: Ubuntu 24.04 LTS
- **Status**: ✅ **RESOLVED**

### **Recommendation**
For anyone facing similar issues:
1. Use the **standard cocotb Makefile structure** from the documentation
2. Let cocotb handle simulator integration automatically
3. Start with simple tests and expand gradually
4. Ensure `cocotb-config` is properly detected

**Issue can be closed as resolved.** The cocotb framework works correctly with Verilator when using the proper approach.

closed 2025-08-01 by shivarammysore