cocotb/cocotb #4848: V2.0.0b1 makefile compatibility issue 2

mornthx, 2025-08-03:
duplicate of #4845 as it was closed.

As @ktbarrett said,  for systems where Python entry points are not automatically added to the user's PATH , 'PYTHON_BIN' falls back on 'python3'. 

The problem is,  V2.0.0b1 does not work like that. 

My python entry point is `python3`, I'm sure`PYTHON_BIN` won't falls back on 'python3' but still 'python'.

mornthx, 2025-08-03:
the current line in makefile.sim:
`PYTHON_BIN ?= python`

I guess the **3** is left behind if it is supposed to be falls back on `python3`.

ktbarrett, 2025-08-03:
You should be able to try this out now by using `pip install cocotb@https://github.com/cocotb/cocotb.git`. As we are in a change freeze this shouldn't change anything else and shouldn't change for the final 2.0 release.