cocotb/cocotb #4866: I use the VCS (Version V-2023.12-SP2_Full64) but get ERROR like this, does new VCS verison unsupported?

JiangfengHong, 2025-08-08:
<pre>(cocotb) [ic@localhost /mnt/hgfs/share/xxx/AXI_Master/sim_1]$make
rm -f results.xml
&quot;make&quot; -f Makefile results.xml
make[1]: 进入目录“/mnt/hgfs/share/xxx/AXI_Master/sim_1”
mkdir -p sim_build
echo &quot;acc+=rw,wn:*&quot; &gt; sim_build/pli.tab
cd sim_build &amp;&amp; \
TOPLEVEL=axi_ram \
/home/jjt/install/synopsys/vcs/vcs/V-2023.12-SP2/bin/vcs -top axi_ram  -debug_access+r+w-memcbk -debug_region+cell +vpi -P pli.tab +define+COCOTB_SIM=1 -sverilog \
-timescale=1us/1us \
-full64 -debug -load /home/jjt/miniconda3/envs/cocotb/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_vcs.so  /mnt/hgfs/share/xxx/AXI_Master/sim_1/../rtl/axi_ram.v

Warning-[DEBUG_DEP] Option will be deprecated
  The option &apos;-debug&apos; will be deprecated in a future release.  Please use 
  &apos;-debug_acc+pp+f+dmptf -debug_region+cell+encrypt&apos; instead.

Warning-[DBGACC_DBG] Multiple debug options being used
  The debug switches &apos;-debug_access&apos; and &apos;-debug*&apos; are being used together.  
  For better performance, consider using only &apos;-debug_access&apos;.

                         Chronologic VCS (TM)
       Version V-2023.12-SP2_Full64 -- Sat Aug  9 00:16:42 2025

                    Copyright (c) 1991 - 2024 Synopsys, Inc.
   This software and the associated documentation are proprietary to Synopsys,
 Inc. This software may only be used in accordance with the terms and conditions
 of a written license agreement with Synopsys, Inc. All other use, reproduction,
   or distribution of this software is strictly prohibited.  Licensed Products
     communicate with Synopsys servers for the purpose of providing software
    updates, detecting software piracy and verifying that customers are using
    Licensed Products in conformity with the applicable License Key for such
  Licensed Products. Synopsys will use information gathered in connection with
    this process to deliver software updates and pursue software pirates and
                                   infringers.

 Inclusivity &amp; Diversity - Visit SolvNetPlus to read the &quot;Synopsys Statement on
            Inclusivity and Diversity&quot; (Refer to article 000036315 at
                        https://solvnetplus.synopsys.com)

     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function &apos;vpi_register_cb&apos; is not supported in the compiler.
  Please fix the code being referred to by the &apos;-load&apos; compile option.

     -.--ns ERROR    gpi                                ..b/vpi/VpiCbHdl.cpp:72   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbStartOfSimulation(11)

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function &apos;vpi_chk_error&apos; is not supported in the compiler.
  Please fix the code being referred to by the &apos;-load&apos; compile option.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function &apos;vpi_register_cb&apos; is not supported in the compiler.
  Please fix the code being referred to by the &apos;-load&apos; compile option.

     -.--ns ERROR    gpi                                ..b/vpi/VpiCbHdl.cpp:72   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbEndOfSimulation(12)

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function &apos;vpi_chk_error&apos; is not supported in the compiler.
  Please fix the code being referred to by the &apos;-load&apos; compile option.

Warning-[DBGACC_CMB] Illegal &apos;-debug_access&apos; usage
  The combination of options &apos;-debug_access+nomemcbk and (-debug_all, -debug, 
  or -debug_pp)&apos; is not valid.  Ignoring the option &apos;-debug_access=nomemcbk&apos;
  Please recompile using the &apos;-debug_access&lt;+options&gt;&apos; switch and incremental 
  options as required.  Recommended options are &apos;-debug_access&apos; for 
  post-process debug, &apos;-debug_access+class&apos; for testbench debug, and 
  &apos;-debug_access+all&apos; for all debug capabilities.  Refer the VCS user guide 
  for more granular options for debug control under the switch &apos;-debug_access&apos;
  and refer to &apos;-debug_region&apos; for region control.

Parsing design file &apos;/mnt/hgfs/share/xxx/AXI_Master/sim_1/../rtl/axi_ram.v&apos;
Top Level Modules:
       axi_ram
TimeScale is 1 ns / 1 ps
Starting vcs inline pass...

1 module and 0 UDP read.
recompiling module axi_ram
make[2]: Entering directory `/mnt/hgfs/share/xxx/AXI_Master/sim_1/sim_build/csrc&apos;
make[2]: Leaving directory `/mnt/hgfs/share/xxx/AXI_Master/sim_1/sim_build/csrc&apos;
make[2]: Entering directory `/mnt/hgfs/share/xxx/AXI_Master/sim_1/sim_build/csrc&apos;
rm -f _cuarc*.so _csrc*.so pre_vcsobj_*.so share_vcsobj_*.so
ln: failed to create symbolic link &apos;_111479_archive_1.so&apos;: Operation not supported
make[3]: *** [_111479_archive_1.so] Error 1
make[2]: *** [product_clean_order] Error 2
make[2]: Leaving directory `/mnt/hgfs/share/xxx/AXI_Master/sim_1/sim_build/csrc&apos;
Make exited with status 2
CPU time: 1.586 seconds to compile + 1.211 seconds to elab + .302 seconds to link
make[1]: *** [sim_build/simv] 错误 2
make[1]: 离开目录“/mnt/hgfs/share/xxx/AXI_Master/sim_1”
make: *** [sim] 错误 2</pre>

aichichayedan, 2025-08-09:
The problem may not be related to Cocotb itself. It's because I'm getting an Error from a window file accessed through a shared folder under linux, and I'm not having a problem copying the project to another directory.
I also noticed that in my version of VCS, specifying GUI=1 directly, Verdi opens with the error：
Error - sim_build/simv.daidir is not generated with the -kdb option.
Specify the -kdb option when generating the database by VCS tools.
By add ：
EXTRA_ARGS += -kdb

It'll works

ktbarrett, 2025-08-29:
@facebreeze VCS 2018 is *very* old. There's little doubt to me it won't work. We can't always support old versions of tools due to the litany of issues they tend to have.

Can you provide the HDL source code as well so we can see if it produces anything in our CI?

ktbarrett, 2025-08-31:
Not sure right now. Apparently loading the VPI during compilation, even though it does nothing, has somw effect. VCS is doing something very weird I haven't figured out yet.