cocotb/cocotb #4868: time precision same as clock causes failure

buzmeg, 2025-08-11:
If I set my time precision with either:

```timeunit 1ns/1ns;```
or
````timescale 1ns/1ns```

And subsequently declare my clock:
```    cocotb.start_soon(Clock(dut.clk, 1, unit="ns").start())```

I get an error of:
```     0.00ns WARNING  ..register_file.test_register_file Failed to start clock: invalid arguments!```

However, if i change the Clock unit to "ms", things work.  If I change the precision to 1ps, things also work.

If I specify:
````timescale 1ms/1ms```

I (properly) get: ``` 0.00ns WARNING  ..register_file.test_register_file Unable to accurately represent 1(ns) with the simulator precision of 1e-3```

So, something appears to be broken if I specify my clock with the exact unit of my timescale precision.

I'm using:
```     0.00ns INFO     cocotb                             Running on Verilator version 5.039 devel
     0.00ns INFO     cocotb                             Seeding Python random module with 1754875810
     0.00ns INFO     cocotb                             Initialized cocotb v2.0.0.dev1+239cad40 from /var/home/foo/work/.venv/lib/python3.10/site-packages/cocotb```

Thanks.

ktbarrett, 2025-08-11:
Yeah that's not possible. There are two transitions of the clock signal in a period. If your clock is 1ns you change the signal every 500ps. If your simulator's timescale is set to 1 ns, you can't represent time steps of 500ps.

Perhaps we can provide a better error message.

buzmeg, 2025-08-11:
Yeah, the "Unable to accurately represent" error would be way better for the other case.  I don't know how hard that would be to make work.

This was a pain because the timeunit declaration snuck in via an imported file and so I was looking all over for a bug when all I really needed to do was get rif of the timeunit declaration.