cocotb/cocotb #4907: Some color logging is not set by Formatter

jontindal, 2025-08-22:
`default_config(color=False)` and `SimLogFormatter(color=False)` do not disable all color logging.

`_log_test_start`, `_record_test_skipped`, `_record_test_init_failed`, `_record_test_passed` `_record_test_failed` and `_log_test_summary` in regressions.py add color based on the value of `want_color_output()`, so cannot be set by a specific logging handler.

The use case for this is to use an additional FileHandler to save logs to a file. I was hoping to set the formatter for this to avoid color in the log file, whilst keeping the default color formatting in the terminal (so preferably avoid using environment variables).

Many thanks,
Jon

ktbarrett, 2025-08-22:
Technically The color is the colorization of the formatter is for the logging prefix only, users can still add arbitrary ansi escape code sequences to messages, which is what the RegressionManager is doing.

That being said we could probably transition the flag to mean "strip ANSI escape codes" using the code that will come in soon from [here](https://github.com/cocotb/cocotb/pull/4900/files#diff-5a7acfa53675d7a1a4a3a313bde5162ec9226182a26b403a652a93acfc2e6cc8R239-R254).

That will be slower than just not creating the ANSI escape codes in the first place by using `COCOTB_ANSI_OUTPUT=0`, which also defaults the `color` argument to the Formatters.

jontindal, 2025-08-22:
No worries, I am using a regex to strip the color at the minute but was just wandering whether if I was missing a better alternative

ktbarrett, 2025-08-22:
Yeah `COCOTB_ANSI_OUTPUT=0` to make `want_color_output()` return `False` is the best way. But there's definitely some functionality we can add here.

I want all envvar flags like this to be runtime switchable, so people aren't forced to use the envvars in other contexts. So we will need to transition to some global flag over `want_color_output` like we did with `cocotb.debug.debug`. And then stripping the ANSI codes out of the messages is something we can build in as a feature to the logger.

ktbarrett, 2025-08-26:
Closed by #4909. Thanks for the report! I think this system is better. Sorry the API shifted a bit there.

closed 2025-08-26 by ktbarrett