cocotb/cocotb #4919: Make quickstart example less "basic"

ktbarrett, 2025-08-27:
Per #4917. The quickstart design is probably to simplistic to be useful to people. We need a design that can start simple then be scaled up to something that can showcase a lot of what cocotb can do. Also be sure to include things like dumping waves to make it familiar to more experienced devs.

georgmagne, 2025-09-10:
Is this up for grabs? I was also a bit frustrated with the quickstart guide when I was starting out a few years back. But I have since gotten more familiar, and would like to help out here as my first issue!

I have started slightly with a more comprehensive example with 8-bit counter that can be enabled/disabled. The testcase uses Clock() and some triggers out of the box. IMO that is a better intro than manually making a clock like the current example.

A few questions, if it is up for grabs:
1. Is the python runner still `experimental and subject to change`? I have switched to it quite some time ago and not looked back since.

2. Anything in particular you would like to see described in the Quickstart?

ktbarrett, 2025-09-10:
Yes absolutely go for it!

Generally looking for
* Python runner
* How to create a test
* How to access signals in DUT
* How to get/set values on signal
* How to use triggers, specifically Timer and (Rising/Falling/Edge)
* How to run things concurrently with `start_soon`
* How to fail a test with `assert`
* How to dump waves and view them with either gtkwave or surfer

At some point introducing Clock is good. Might be a good idea to show manually writing one then introducing the built-in one in the next step and emphasize "reuse" and "abstraction".

ktbarrett, 2025-09-10:
> Is the python runner still experimental and subject to change? I have switched to it quite some time ago and not looked back since.

Yes and no. It still isnt at feature parity with the Makefiles, but its getting closer. I'd be comfortable with that being the way the quickstart example works.